arduino/ArduinoCore-arc32 #89: txPin and rxPin are defined as global variables in SoftwareSerial.cpp

soundanalogous, 2015-12-19:
Because the variables `txPin` and `rxPin` are defined as global variables in SoftwareSerial.cpp (`/CurieSoftwareSerial/SoftwareSerial.cpp`) they cannot be used in sketches. This is going to break a lot of existing sketches since they are commonly used variable names in sketches. Please make these variables private or prepend an underscore (_txPin) or other convention to differentiate.

soundanalogous, 2015-12-19:
Looks like `txPin` is no longer defined globally but `rxPin` still is: https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieSoftwareSerial/SoftwareSerial.cpp#L49

soundanalogous, 2015-12-19:
Also, I don't see any reason why `rxPin` can't be defined as a private property in SoftwareSerial.h because it is not ever used in the global scope. All use is within methods.

rwaldron, 2015-12-19:
> defined as global variables in SoftwareSerial.cpp 

I'd go further and require that this implementation must only expose those identifiers that the [SoftwareSerial](https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/libraries/SoftwareSerial) exposes.

kitsunami, 2016-02-03:
Please review. If not fixed, assign to Sidney.

soundanalogous, 2016-02-03:
It was fixed: https://github.com/01org/corelibs-arduino101/pull/101#issuecomment-173016516.

jgithubs, 2016-02-08:
Verified as fixed on 2/8/16

closed 2016-02-10 by kitsunami