arduino/ArduinoCore-arc32 #95: [CurieBle] - Make connection interval configurable

soundanalogous, 2016-01-04:
Currently the connection interval is hard-coded in ble_client.c to a [min value of 80ms and a max value of 150ms](https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieBle/src/internal/ble_client.c#L60-L61).

I have found by logging (via PacketLogger) and changing the min and max connection values in ble_client.c that the final interval that gets set seems to always match the max interval of 150ms. This means that that fastest a user can read an analog input or blink an LED for example is at an interval of 150ms which is a bit long.

Is there a specific reason that the connection interval range is set to 80ms = 150ms?

I've tried a much lower range [7.5ms / 30ms] and did not encounter any issues. This would enable me to read or write at a frequency of 30ms (since it seems MAX is always used). I understand this may be too fast (and thus consume too much battery) for some applications, but for others (if battery is less of an issue vs responsiveness of a physical UI) it makes sense to have a faster interval.

What would be idea is for the user to be able to set the interval range. It could default to 80 / 150 as it currently does, but could be overridden. A set of absolute min / max would still need to be applied behind the scenes but it would at least enable the user to have a choice per their particular application.

soundanalogous, 2016-01-04:
According to the nRF51822 documentation the MIN and MAX connection interval the driver can handle is 7.5ms and 4s respectively: https://devzone.nordicsemi.com/documentation/nrf51/4.1.0/html/group___b_l_e___g_a_p___c_p___l_i_m_i_t_s.html#gac6071964ea94ef548da1beaf23f032da.

monteslu, 2016-01-26:
Agree 100%

Please allow this to be configurable.

soundanalogous, 2016-01-30:
@sandeepmistry do you have time to take a look at this within the next couple of weeks? If not could you point me to the code you mentioned a while back on gitter for setting the advertising the advertising interval and I'll try to take a pass at it. I think this is something you had implemented in arduino-BlePeripheral and mentioned it could be a good pattern to apply to CureBle for specifying the connection interval.

kitsunami, 2016-02-24:
@bigdinotech please reach out to @sandeepmistry to see where we are on this.

sandeepmistry, 2016-02-24:
@kitsunami @bigdinotech I've submitted #114 to resolved this.

soundanalogous, 2016-03-01:
This was merged yesterday: https://github.com/01org/corelibs-arduino101/pull/114.

closed 2016-03-01 by soundanalogous