arduino/ArduinoCore-arc32 #97: Functions "*.disconnect()" and "*.end()" doesn't work

mwms, 2016-01-05:
example
ble_peripheral.end()
central.disconnect()

kitsunami, 2016-02-24:
To Sidney, to be worked on with JIRA 544.

SidLeung, 2016-05-27:
Please note that the current Arduino platform can only be a Peripheral, no Central features are available.  If you are trying to finding out if your board is still connected to the host, you can use the method, BLECentral::connected().

sandeepmistry, 2016-05-27:
@SidLeung I looked into this a few months ago, the issue is ```central.disconnect()``` does not trigger a disconnection of the connected central. From what I remember the handle book keeping in `CurieBLE` looked ok, I suspect there is something wrong on the nRF51 f/w side or incorrect messaging format.

SidLeung, 2016-05-27:
@sandeepmistry Thanks for the info.  Will track that down,

ccoulton, 2016-08-16:
Related Question, is there a way to start advertising with an interrupt? end seems to work when not connected to a central, but restarting the perhipheral seems impossiable.

eriknyquist, 2016-09-29:
@SidLeung any idea if this has been fixed in the recent BLE merge?

noelpaz, 2016-10-11:
I verified that disconnect and end works from a peripheral using the API disconnect and end for the current CurieBLE implemented by @sgbihu  (Liang's team) . I am closing this.

Thanks

closed 2016-10-11 by noelpaz