arduino/ArduinoCore-arc32 #112: No I2c slave mode support in wire.h

francisc0garcia, 2016-02-01:
the callbacks

> Wire.onReceive(receiveData);

and

> Wire.onRequest(sendData)

 are missing compare to previous versions. Now it is not possible to setup arduino 101 as slave to enable communications with other devices such as Raspberry or similar

Some overloads for method "begin" are also missing 

> error: no matching function for call to 'TwoWire::begin(int)'

Dan-Emutex, 2016-02-02:
@francisc0garcia Arduino 101 does not support I2C Slave Mode at the hardware level - only Master Mode is supported.  That is why those callbacks are missing.

francisc0garcia, 2016-02-09:
Thank you for explanation!

closed 2016-02-09 by francisc0garcia

rei-vilo, 2016-07-01:
Any plan to implement a software-based slave I²C on Arduino / Genuino 101?

SidLeung, 2016-07-01:
We have examined the means to implement a software I2C.  However, it was determined that a follow on product with an actual hardware I2C interface made a lot more sense.

Pachicito, 2017-04-10:
Why? Why is I2C slave not supported on the 101?

SidLeung, 2017-04-10:
The I2C hardware interfaces originate from the ARC core (the processor used by the Arduino platform) of the Curie SoC only support Master mode.  Please refer to the Curie data sheet for more details.