arduino/ArduinoCore-arc32 #130: Error when using latest version from git on linux following the instructions

arfoll, 2016-03-02:
I put this in the place of the 1.0.4/ board folder and when I hit 'verify' I get this:

```Board arduino_101 (platform arc32, package Intel) is unknown

Error compiling.```

Not really too sure what I'm doing wrong :/

bbaltz505, 2016-03-02:
The zip actually contains a top-level folder (corelibs-arduino101-master). We should be extracting the contents of that folder into Arduino15\packages\Intel\hardware\arc32\<version>\

arfoll, 2016-03-02:
I did use the zip but cloned from git... Directly into that folder (it
didn't create a subfolder)

bbaltz505, 2016-03-02:
If you clone the repo without any arguments, it will create a corelibs-arduino101 folder within 1.0.4/. I cloned the repo to "." and the IDE compiled ok.
`cd 1.0.4`
`git clone https://github.com/01org/corelibs-arduino101.git .`
So, as long as the contents (libraries, platform.txt, etc.) of the repo are in the 1.0.4 folder, things should work.

arfoll, 2016-03-02:
I added the '.'.. I'll double check tmrw but my guess is there's something
else. Are you using the nightly arduino build or the latest stable?

bbaltz505, 2016-03-02:
I'm using the 1.6.7 IDE release

pricopb, 2016-03-03:
@arfoll, do you have any other directories, except 1.0.4, in the ".arduino15/packages/Intel-Test/hardware/arc32" directory?
If you do, please delete them and try again.

arfoll, 2016-03-03:
Indeed @pricopb I'd kept the old folder as backup named `dudeman` :). I made a PR to improve the instructions

bbaltz505, 2016-03-03:
Thanks Brendan. We'll add the note. Are you using a nightly IDE build? I'm told that this is a known issue but I don't know if it has been fixed yet.

arfoll, 2016-03-03:
No this was on the release 1.6.7 one on 64bit linux.

closed 2016-03-03 by arfoll

deadprogram, 2016-03-05:
Hi, @arfoll & @bbaltz505 

I have a similar problem with the 1.6.7 release on 64bit Linux. However, in my case I do not have any extra folders in my `~/.arduino15/packages/Intel/hardware/arc32` directory.

My error message is:

```
java.io.IOException: Cannot run program "{runtime.tools.arduino101load.path}/arduino101load/arduino101load": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:123)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:167)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1178)
	at processing.app.Sketch.exportApplet(Sketch.java:1152)
	at processing.app.Sketch.exportApplet(Sketch.java:1124)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2430)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
```

I followed the same procedure to do the in-place manual install. Able to compile, but not upload.

calvinatintel, 2016-03-05:
@deadprogram The latest tools were updated to be incompatible with past releases. I recommend that you update both the corelibs and the tools at the same time.

deadprogram, 2016-03-05:
@calvinatintel so I need latest nightly build of Arduino?

calvinatintel, 2016-03-05:
@deadprogram Here is the tools: https://github.com/01org/intel-arduino-tools. Replace `~/Library/Arduino15/packages/Inte/tools/arduino101load/<version>`.

deadprogram, 2016-03-05:
Thanks, @calvinatintel 

Any particular version of Golang needed to compile that utility?

calvinatintel, 2016-03-05:
@deadprogram We just used the latest one. The precompiled version should work out of box though. Be sure to copy the correct branch for your OS.

deadprogram, 2016-03-05:
Was able to compile, then copied the file to `~/.arduino15/packages/Intel/tools/arduino101load/1.0.4/arduino101load`

but still seems like the `runtime.tools.arduino101load.path` is not set correctly?

soundanalogous, 2016-03-05:
Same issue on OS X. Something is missing beyond adding `~/Library/Arduino15/packages/Intel/tools/arduino101load/<version>/arduino101load/arduino101load`

I get this error:
```
java.io.IOException: Cannot run program "{runtime.tools.arduino101load.path}/arduino101load/arduino101load": error=2, No such file or directory
...
```

The path arduino101load director does not exist within tools. I tried adding within the existing `sketchUploader` directory (`sketchUploader/<version>/arduino101load/`) but that does not work either. I suspect there is an additional step that the installer runs (when updating via Boards Manager) that is not included when copying the tools and curie-101-core alone.

StevenShuoFeng, 2016-03-05:
@soundanalogous 
These tools are not available when you install the 'Intel Curie Boards' library from Arduino IDE. However, the platform.txt under `Library/Arduino15/packages/Intel/hardware/arc32/1.0.4/` in this repo is using 'arduino101load' in the bottom compile patterns part.

I solved this problem by replacing the platform.txt with the old one at the same location:

1. Delete the Curie library and install again. 
2. Save a copy of the platform.txt
3. Update the 'Library/Arduino15/packages/Intel/hardware/arc32/1.0.4' directory using this repo. 
4. Replace the platform.txt from 2.

soundanalogous, 2016-03-05:
@StevenShuoFeng works great, thanks!

bbaltz505, 2016-03-06:
@deadprogram, we use Go version go1.5.3. Older versions may have issues. I know I had problems with the default version on Ubuntu 12.04 (go1)