arduino/ArduinoCore-arc32 #143: Klocwork issue in stdlib_noniso.cpp

bbaltz505, 2016-03-08:
In cores/arduino/stdlib_noniso.cpp (line 153):

function 'sprintf' possibly accepts format string that may be influenced by user, causing format string vulnerability. Undefined string lengths can lead to buffer overflows and potential exploitation by attackers. Use a defined value for string lengths.

```
char *dtostrf (double val, signed char width, unsigned char prec, char *sout) {
   ...
   sprintf(sout, fmt, val);
```

bbaltz505, 2016-03-11:
Hi @facchinm can you take a look at this? We do static code analysis of corelibs code with a tool called Klocwork and it flagged this potential issue in stdlib_noniso.cpp. Does this look serious enough to fix or should we ignore it? See [this page](https://developer.klocwork.com/documentation/en/insight/10-1/sv-fmtstr-generic) for info on the defect type.

facchinm, 2016-03-14:
Hi Brian,
as defined in AVR libc manual [here] (http://www.atmel.com/webdoc/AVRLibcReferenceManual/group__avr__stdlib_1ga060c998e77fb5fc0d3168b3ce8771d42.html), the user is responsible for providing a char buffer big enough to contain the string, so IMO the vulnerability should not be considered a threat :smile:

bbaltz505, 2016-03-14:
Thanks. I'll ignore future defects of this type.

closed 2016-03-14 by bbaltz505