arduino/ArduinoCore-arc32 #155: Zero Motion Detection Example Sketch

mjs513, 2016-04-01:
Recommend a example sketch be provided for zero motion detection using the CurieIMU.  Would be advantages to for trying to get odometry implemented.

SidLeung, 2016-06-01:
New IMU example Sketches for Free Fall, Zero Motion are provided in the Bellatrix release.

mjs513, 2016-06-01:
Thank you.  I will give a try as soon as I get some time.  Was having a devil of a time getting it to work properly

mjs513, 2016-06-02:
Ok.  Just tried to compile the ZeroMotionDetect and  MotionDetect sketches and received the following error messages on compile:

arc-elf32-g++: error: unrecognized argument in option '-mcpu=quarkse_em'

arc-elf32-g++: note: valid arguments to '-mcpu=' are: ARC600 ARC601 ARC700 ARCv2EM ARCv2HS arc600 arc601 arc700 arcem archs

bbaltz505, 2016-06-02:
Ah, the latest corelibs requires a newer toolchain:
https://downloadmirror.intel.com/25470/eng/arc-toolchain-windows-arcem-1.0.1.zip (Windows)
https://downloadmirror.intel.com/25470/eng/arc-toolchain-osx-arcem-1.0.1.tar.bz2 (OSX)
https://downloadmirror.intel.com/25470/eng/arc-toolchain-linux64-arcem-1.0.1.tar.bz2 (Linux64)

Extract one of these to replace your toolchain (compiler) files in .arduino15/packages/Intel/tools/arc-elf32/1.6.2+1.0

mjs513, 2016-06-02:
Will give it a try.  Just to double check the toolchain is 1.6.2+1.0 replacing 1.6.4+1.0: going backwards in the numbering?????

mjs513, 2016-06-02:
hi @bbaltz505.  Just gave it a try and seems to work.  There does seem to be a lag between moving the board and when the interrupt is tripped.  It gets better as I reduce values.

Mike

bbaltz505, 2016-06-20:
hi @mjs513 . Have you tried this issue with the latest release (1.0.6)?

mjs513, 2016-06-24:
Hi @bbaltz505 haven't tried it with the official release as of yet.  Been busy with my rover project and family issues.  Will give it another try.  Think the real problem is with the algorithm itself which is not an arduino 101 issue.  If I move the board slowly does not really register motion at least with the initial tests I made..  I have another method that implements a Zara/Zupt algorithm that works fairly well.

Mike

mjs513, 2016-06-26:
@bbaltz505 I got around to trying the examples for both zero motion and motion detect.  I think that should close this issue.  I have not checked if issue 181 would affect the performance but that is another issue.  Just as a note setting the right thresholds and durations can be a pain.  

 I really like the performance of the Motion Detect Example better than Zero Motion.  Thanks for incorporating the examples should be a help to everyone.

Mike

closed 2016-06-26 by mjs513