arduino/ArduinoCore-arc32 #158: Potential Conflict between c++ vector class and CurieBLE

mjs513, 2016-04-05:
I am trying to run a sketch using the c++ lib vector (#include <vector>) but there seems to be a conflict with CurieBLE.  The sketch works fine without the include <vector> .  Any suggestions would be appreciated.

Here is a list of the error messages:

```
Arduino: 1.6.7 (Windows 10), TD: 1.27, Board: "Arduino/Genuino 101"

C:\Users\CyberPalin\Documents\Arduino\Arduino101\Arduino101_CRP_v2\Arduino101_CRP_v2\Arduino101_CRP_v2.ino: In function 'void setup()':

Arduino101_CRP_v2:71: error: 'blePeripheralConnectHandler' was not declared in this scope

   blePeripheral.setEventHandler(BLEConnected, blePeripheralConnectHandler);

                                               ^

Arduino101_CRP_v2:72: error: 'blePeripheralDisconnectHandler' was not declared in this scope

   blePeripheral.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);

                                                  ^

Arduino101_CRP_v2:75: error: 'rxCharacteristicWritten' was not declared in this scope

   rxCharacteristic.setEventHandler(BLEWritten, rxCharacteristicWritten);

                                                ^

C:\Users\CyberPalin\Documents\Arduino\Arduino101\Arduino101_CRP_v2\Arduino101_CRP_v2\Arduino101_CRP_v2.ino: In function 'void loop()':

Arduino101_CRP_v2:148: error: 'closeConnection' was not declared in this scope

       closeConnection(parser.getPfodAppStream());

                                                ^

exit status 1
'blePeripheralConnectHandler' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

```
I posted this on the arduino101 forum but no takers. The sketch can be downloaded from the [forum](Conflict between Include <vector> and CurieBLE.h)

Thanks
Mike

mjs513, 2016-04-11:
Update:  I downloaded the new Arduino-builder that should have fixed the issue per faccihm however it still threw the same error.  I also used the new builder with an old sketch that worked fine (no vector or strange calls) and it balked at the following:

In main sketch I do a call like so:
`
uint8_t count = serial_busy_wait();`

It states that serial_busy_wait() is not defined in this context or something like that.  Here is the function itself:

```
char serial_busy_wait() {
  while(!Serial.available()) {
    ; // do nothing until ready
  }
  return Serial.read();
}
```

mjs513, 2016-06-24:
Just out of curiosity is there any status on this and what the issue is?

eriknyquist, 2016-06-24:
Hi @mjs513, the link you posted is actually this: https://github.com/01org/corelibs-arduino101/issues/Conflict%20between%20Include%20%3Cvector%3E%20and%20CurieBLE.h

Can you post a link to the forum post you talked about?

mjs513, 2016-06-24:
Believe this is the correct link.

https://forum.arduino.cc/index.php?topic=391375.0

Mike

eriknyquist, 2016-06-24:
@mjs513 I downloaded your sketch and compiled it, and I still see those errors with `#include<vector>` in or not, not really sure why that made the errors go away for you.

So, about the errors; in C++, a function has to be defined or declared before it can be called. You're calling a bunch of functions in `setup()` that haven't been declared or defined yet. The first thing to do is make sure all the functions you use in `setup()` are defined before they get called.

I moved the definitions for `blePeripheralConnectHandler` and `rxCharacteristicWritten` up before `setup()` and tried to compile, just to make sure there wasn't something else going on. The original error messages went away, but there were some new ones about more functions not defined and a couple of syntax errors. Can you fix those problems, and see if it works for you then?

mjs513, 2016-06-24:
@eriknyquist I am using IDE 1.6.7 for reference but I just installed 1.6.9 to keep things on the same footing.  So before rewriting the program I posted I decided to do a simple test.  I copied an example script on using vectors in c++ and it ran fine in both 1.6.7 and 1.6.9.  Next I used the EXAMPLE BatterMonitor sketch and merged the two to see what would happen since the example sketch is a relatively simple example of using CurieBLE.  It did not work.  Here is the error message:

> vector-test:101: error: 'updateBatteryLevel' was not declared in this scope
>          updateBatteryLevel();
>                             ^
> exit status 1
> 'updateBatteryLevel' was not declared in this scoped

Here is the sketch:
```
#include <CurieBLE.h>
#include <vector>
#include <Streaming.h>
#include <Wire.h>
#define cout Serial

using namespace std;

/*  */
BLEPeripheral blePeripheral;       // BLE Peripheral Device (the board you're programming)
BLEService batteryService("180F"); // BLE Battery Service

// BLE Battery Level Characteristic"
BLEUnsignedCharCharacteristic batteryLevelChar("2A19",  // standard 16-bit characteristic UUID
    BLERead | BLENotify);     // remote clients will be able to
// get notifications if this characteristic changes

int oldBatteryLevel = 0;  // last battery level reading from analog input
long previousMillis = 0;  // last time the battery level was checked, in ms

int icount;

void setup() {
    cout.begin(9600);
  pinMode(13, OUTPUT);   // initialize

 /* Set a local name for the BLE device
     This name will appear in advertising packets
     and can be used by remote devices to identify this BLE device
     The name can be changed but maybe be truncated based on space left in advertisement packet */
  blePeripheral.setLocalName("BatteryMonitorSketch");
  blePeripheral.setAdvertisedServiceUuid(batteryService.uuid());  // add the service UUID
  blePeripheral.addAttribute(batteryService);   // Add the BLE Battery service
  blePeripheral.addAttribute(batteryLevelChar); // add the battery level characteristic
  batteryLevelChar.setValue(oldBatteryLevel);   // initial value for this characteristic

  /* Now activate the BLE device.  It will start continuously transmitting BLE
     advertising packets and will be visible to remote BLE central devices
     until it receives a new connection */
  blePeripheral.begin();
  Serial.println("Bluetooth device active, waiting for connections...");

}

void loop() {
    vector <int> example;         //Vector to store integers
    example.push_back(3);         //Add 3 onto the vector
    example.push_back(10);        //Add 10 to the end
    example.push_back(33);        //Add 33 to the end
        
    cout<<"---EXAMPLE----" << endl;
    icount = example.size();
    for(int x=0; x<icount; x++) 
    {
        cout<<example[x]<<" " << endl;    //Should output: 3 10 33
    }
    if(!example.empty())          //Checks if empty
        example.clear();          //Clears vector
    vector <int> another_vector;  //Creates another vector to store integers
    
    another_vector.push_back(10); //Adds to end of vector
    example.push_back(10);        //Same

    if(example==another_vector)   //To show testing equality
    {
        example.push_back(20); 
    }
    
    icount = example.size();
    cout<<"---ANOTHER EXAMPLE----" << endl << endl;
    for(int y = 0; y < icount; y++)
    {
        cout<<example[y]<<" " << endl;    //Should output 10 20
    }
    another_vector.clear();
    example.clear();

 // listen for BLE peripherals to connect:
  BLECentral central = blePeripheral.central();

  // if a central is connected to peripheral:
  if (central) {
    Serial.print("Connected to central: ");
    // print the central's MAC address:
    Serial.println(central.address());
    // turn on the LED to indicate the connection:
    digitalWrite(13, HIGH);

    // check the battery level every 200ms
    // as long as the central is still connected:
    while (central.connected()) {
      long currentMillis = millis();
      // if 200ms have passed, check the battery level:
      if (currentMillis - previousMillis >= 200) {
        previousMillis = currentMillis;
        updateBatteryLevel();
      }
    }
    // when the central disconnects, turn off the LED:
    digitalWrite(13, LOW);
    Serial.print("Disconnected from central: ");
    Serial.println(central.address());
  }
}
```

The Streaming.h lib is available here: http://arduiniana.org/libraries/streaming/

I am not sure how you got it working with 1.6.9.  You can give this a try.
Mike

eriknyquist, 2016-06-24:
You are calling **`updateBatteryLevel();`** but there is no definition for it. Again, in C++, a function has to be defined or declared before you can use it.

mjs513, 2016-06-24:
This is your example directly from the 101 examples provided with corelib.
The is no updatebatterylevel function in the example.

eriknyquist, 2016-06-24:
@mjs513 there is indeed a definition for the updateBetteryLevel function, both in the latest version on github and on the released version that you download through the IDE boards manager. Perhaps you removed it and saved it by accident?

However, I see that that function is actually called before it is defined, and it compiles fine. If I add the <vector> include, then I get the "not declared in scope" error. Then, finally, if I leave in the <vector> include and move the function definition up before it is called, then it compiles again. This is very strange. I'll try and find out what's going on.

mjs513, 2016-06-24:
I will double check the function call. Typically i put all my functions
after the loop or in another tab. They always compile without a problem.
This is even with the 101. The issue seems to be when i add the include for
vector.

Thanks for checking.

and it compiles fine. If I add the include, then I get the "not declared in
scope" error. Then, finally, if I leave in the include and move the

eriknyquist, 2016-06-24:
OK so it turns out that while C++ does require functions to be defined/declared before being called, the Arduino builder gets around this by doing a first pass over your sketch and writing your function declarations for you. So, I was wrong about that-- you don't need to move your function definitions.

So the question now is, why does adding `#include <vector>` break that? Stay tuned to find out...

mjs513, 2016-06-26:
That is the million dollar question.  I have been staying tuned since I posted this issue back in April.  I believed that this maybe related to the question you asked me on the forum page.

mjs513, 2016-07-05:
Just tested nightly build for 1.6.10 and the conflict between vector and ble seems to be resolved.  Going to go ahead and close this issue unless further issues arise.  Thanks for fixing the problem.

Mike

closed 2016-07-05 by mjs513

eriknyquist, 2016-07-05:
Thanks @mjs513, but we didn't actually do anything, the Arduino guys just got that fix into their nightly builds by the sounds of it. Glad it's all working now, anyway!