arduino/ArduinoCore-arc32 #166: Need a setClock function in Wire Library

mjs513, 2016-04-14:
To effectively use the i2c bus users need a way to programmatically change the I2C bus speed.  Currently this is implemented for most of the Arduino boards like the mega as well as the teesnsy.  Many sensors that would be connected to the bus would benefit from this ability.

bigdinotech, 2016-04-20:
Hi, I added the functionality to change the I2C clock speed. The PR is under review.

Meanwhile, you can try it out here: 
https://github.com/bigdinotech/corelibs-arduino101/tree/i2cclock

mjs513, 2016-04-21:
Hi bigdinotech
I downloaded the updates per the instructions.  I decided to run the test on the new Teensy Prop Shield that has two I2C sensors from NXP (FXAS21002C and the FXOS8700CQ).  Running 1024 sensor reads using both sensors here is what I got (average of 1024 reads);

slow = 69 milliseconds 
fast = 68 milliseconds

Not much of a difference the bus speed change.   Do you have a test case I can try.  I will give it a try tomorrow using a different sensor suite.

Here is the sketch for your reference:
`#include <Wire.h>
#include <FXAS21002Cnxp.h>
#include <FXOS8700CQnxp.h>

FXOS8700CQ sensor = FXOS8700CQ(); 
FXAS21002C sensor2 = FXAS21002C(); 

unsigned long start, stop;
  int ax, ay, az;
  int gx, gy, gz;
  int mx, my, mz;

void setup() {
  Serial.begin(115200);
  Wire.begin(I2C_SPEED_SLOW);
  
    // Initialize the FXOS8700CQ
  sensor.begin();
  
  // Initialize the FXAS21002C
  sensor2.begin();
  delay(6000);
}

void loop() {
  Serial.println("Testing raw reading speed (average on 1024 samples):");
  start = micros();
  
  for(int i=0; i<1024; i++) {
    // Read the motion sensors
    sensor.readMotionSensor(ax, ay, az, mx, my, mz);
    sensor2.readGyro(gx, gy, gz);
  }
  
  stop = micros();
  Serial.print("--> result: ");
  Serial.print((stop - start) / 1024);
  Serial.print(" microseconds .... ");
  Serial.print(((stop - start) / 1024) / 1000);
  Serial.println(" milliseconds");    

  Serial.println("Looping again..");
  Serial.println("----");
}
`

mjs513, 2016-04-21:
Hi.  Hooked up a MPU-6050 sensor to the mega and 101 here are the results:
                 100hz          400hz
Mega         0.426           0.178 milliseconds
101            4.28             3.29   milliseconds.

all I can say is something does not make sense with the 101 and the I2C implementation
Hope this info helps
Mike.

bigdinotech, 2016-04-21:
@mjs513 can you give a link to the library you are using?
I have hunch that Wire.begin() is being called inside that library and thus overriding the speed back to 100Khz

mjs513, 2016-04-21:
@bigdinotech I did too so I looked at all libs used and deleted the wire.begin's in the [FXOS8700CQ ](https://github.com/mlwarner/fxos8700cq-arduino.git)and the  [FXAS21002C ](https://github.com/sabas1080/FXAS21002C_Arduino_Library.git)libraries.  There are calls to wire.begintransmission(addr) etc. I also looked at the [I2Cdev lib and the MPU60xx ](https://github.com/mjs513/FreeIMU-Updates/tree/master/libraries)lib that I am using from Jeff Rowberg  and there are no wire.begin calls in the libs or anything strange to reset the TWBRs (but that would have a thrown an error).  

I also just hooked up the FDRM-STBC-AGM01 shield from Freescale which have the same chip set as the prop shield with the Curie and got the same results.  However, when I first hooked it up I had the sketch already loaded and it gave me a 2 ms reading for slow.  When I changed the setting to fast it went bac to around 68 ms and when I put it back to slow it gave me 68ms.  Not sure what is happening in this case.

Mike

bigdinotech, 2016-04-21:
@mjs513 I just tried the example sketch from the FXAS21002C library.
I have a scoped attached to SCL and I can see the clock getting faster as expected.
Do you have access to a scope? What clock speed are you getting on the SCL pin?

mjs513, 2016-04-22:
@bigdinotech I have access to a usb scope but now can't get a nice clean signal but from what I am seeing at I2CSLOW the freq varies from 0.058 kHz to 0.086 kHz using a 20ms time div. At I2CFast I get a pretty steady reading of 0.058 kHz using the same settings.  Go figure.  I am attaching the screenshots so you can interpret for me - I really have no idea what I am doing with the scope yet.
Thanks for your help
Fast
[image]

Slow shot 1
[image]

Slow shot2
[image]

bigdinotech, 2016-04-22:
@mjs513 I think if you zoom in more into the signal you would see that the clock speed is correct.

mjs513, 2016-04-22:
@bigdinotech  I believe you.  Like I said I am not very good with a scope.  Anyway so if the clock speed has increased (I am assuming slow is 100 and Fast is 400) why such a difference in the sensor read timing tests between the mega and the Curie?  Why no change in the timings between slow and fast on the curie? Is this a separate issue or am I doing something wrong?

Thanks for your help 
Mike

mjs513, 2016-04-25:
@bigdinotech Would it be possible to add a function that does the same thing called Wire.setClock(xxxxxxx); - this would make it similar to existing calls to setClock in the Arduino wire library.  The function would just do a wire.begin(I2C_SPEED_FAST) or wire.begin(I2C_SPEED_SLOW) if xxxxxx is 4000000 or 4000000L.  It would avoid the user having to change existing code if they are moving a sketch from a mega or uno over to the Cuire.

bigdinotech, 2016-04-25:
@mjs513. I'll take a look at it

mjs513, 2016-04-25:
@bigdinotech Thank you.  I hate keep testing for the board in the code, there are getting to be too many.  I think something simple like in wire.cpp:

```
void TwoWire::setClock(int speed){
        if(speed == 4000000) {
             wire.begin(I2C_SPEED_FAST);
        } else {
             wire.begin();
        }
}

```
I still do not understand the differences in the read speeds between the curie and the mega.

piroozn, 2016-06-23:
I don’t have the sensor you are using but I can test this using I2C RGB display board from my Grove kit plus. I will do that next week and let you know.
I think if you zoom in with scope and look at the clock frequency for the Atmel mega, you will find that it is running slower then 400khz and that may be the reason it is running faster. Slower communication without error can result in faster communication since it does not have to timeout and wait to recover from errors. Also mega firmware may be enabling their internal pullup.
Arduino 101 board does not have any pullup for the I2C. You need to enable the internal pullup for these pins since there are voltage translator used on Arduino 101 and add pullup on your sensor board for these pins. If you are using Grove kit plus, they have place to install these on their board. 
The I2C 0 and UART1 are sharing the pins on the connector. Make sure you are not enabling or using UART 1 since they both use the same pins on Arduino 101.
Looking at your scope picture, I see the voltage reading is about 3.3v. Looks like you don’t have any pullup on the sensor board. Where are you measuring this signal?

mjs513, 2016-06-24:
Hi @piroozn.  To answer some of your questions I am measuring the signal at the board pins while the sensors while the program is running.  The 6050 board is a GY-521.  It looks like the board has 4.7k pullups on the board already (attached), same for the propshield which is 3.3v powered while the 521 is powered from the 5v line.  On the 521 this is LDO to convert from 5v to 3.3v. I am attaching the schematics for your reference.

I believe you are correct that the mega has the internal pull-ups enabled by default (as well as the Uno).  So how to I enable them on the Arduino 101? I know how to disable them on the Mega and Uno. What size additional pullups would you recommend?

As for the UART1 and I2C0 conflict - there should be no problem.  I am only using UART0.  This may be the reason I was having a issue when when I was using an xbee on UART1 and the I2C.  

Thanks for your help.  There does seem to be alot of questions out there with the I2C bus on the 101 board.  By the way if you want one of the sensors send me an address and I can send it to you (CyberMerln@gmail.com).

Mike

mjs513, 2016-06-24:
I just came across this on the forum that might be of interest.

https://forum.arduino.cc/index.php?topic=406458.0

bigdinotech, 2016-06-24:
@mjs513 
Can you try out:
https://github.com/bigdinotech/corelibs-arduino101/tree/i2cdelay
It includes the reduced delays for i2c and the ability to change to clock speed.
It based on Arduino101 corelibs 1.06 so you may have to update your core first.

piroozn, 2016-06-24:
Hi @mjs513

-        The 4.7k pullup for 6050 and 2.2k pullup for the AXIS boards are ok.

-        The interface board is using 3.3v for its VCC and Arduino 101 needs 5V logic

-        For internal pullup, look at the following link. pinMode(2,INPUT_PULLUP); https://www.arduino.cc/en/Tutorial/DigitalPins

-        On the Arduino 101, We have level shifter. Take a look at the data sheet for it and you see that require a pullup on its input and pullup on its output. You have pullup on your sensor board and need pullup on its input. You can enable Curie internal pullup for this.

-        Even you are not directly using UART 1. Make sure the no other OS code is printing error messages to it. Connect a terminal to it without the I2C connection and run your code to see if anyone is printing to the UART just as a test. I remember that some of the BLE code can optionally print messages on that port and accept commands.

-        I will test and debug this next week with the RGB display board or your board if you send me one.

-        Also send me the complete code for this test. I noticed you are missing some of the include statement. I have Arduino Uno and Arduino 101 so I can test both. I have done some performance test between Uno and 101 and notice 101 is lot faster than Uno on most of the software operation.

My address:
2913 Montell ct
Plano Texas 75025

Regards,
Pirooz Najafi

Thanks for your help. There does seem to be alot of questions out there with the I2C bus on the 101 board. By the way if you want one of the sensors send me an address and I can send it to you (CyberMerln@gmail.com<mailto:CyberMerln@gmail.com>).

Mike

intel-larry, 2016-06-25:
Hi Pirooz,

Actually the Arduino 101 is a 3.3v level board. The level shifters are used as protection devices, rather than level shifters. The board only produces 3.3v output. The buffers are used to protect the Curie from accidental exposure to 5 volts.

It only needs (and only produces) 3.3v signals, but 5v signals are perfectly save with it.

Larry Barras
I make stuff go.

PaulStoffregen, 2016-06-25:
Both of these motion sensor chips are 3.3V devices.  They can't handle 5V either.

mjs513, 2016-06-25:
I actually have the FRDM-STBC-AGM01 Sensor Shield from freescale that uses the same chipset as the propshield and seeing the same issues. Interesting that you are saying there is a level shifter. It was my understanding from the board description and a forum question was that the pins are 3.3v but 5v tolerant?  

So for the propshield I would need a level shifter on sda/scl to get it up to 5v? Believe the 2560 this would have to be the case: [](http://playground.arduino.cc/Main/I2CBi-directionalLevelShifter). I am getting very confused with the 101.

I will test the UART1 but for these tests not using ble.  I will get the board out to you either tomorrow or early next week.  I have been running back and forth to the hospital with my mother and she is due home on Monday (been a month).

intel-larry, 2016-06-25:
Yes, the Arduino 101 uses a particular type of level buffer that is automatic bi-directional and it uses it in a way that protects the Curie from over-voltage damage when exposed to 5v levels on its inputs.  

The pins on the Arduino 101 will only produce a maximum logic level of 3.3 volts in output mode. It will not produce 5 volt output though. This is generally compatible with practically all 3.3v and 5v devices. 

When the Curie produces a logic level 0, the buffer will protect the Curie if there is an external pullup to +5 and within the current limits of the device and Curie, pull it down to 0 volts. 

In input mode, the Curie is high impedance (unless you have input_pullup selected for pinMode, in which case it is weakly pulled up).  The buffer will safely accept up to 5 volts input on the Arduino 101 header pin and clamp it to 3.3v on the Curie side of the buffer, thus protecting it. 

We tested a lot of edge cases with pullup and external pullup and floating conditions, and think it works in all cases. 

Usually, i2c relies on external pull-up and "open collector" type drivers to do a pull-down against the resistor. There may be some edge case of voltages/pullups that don't behave well, but we didn't see them in testing and development. 

I2C can be finicky, even as simple as it should be in theory. I2C devices have contributed to many of my grey hairs.

mjs513, 2016-06-25:
Thanks for the info Larry - I2C has been driving me crazy as well.  Sorry I missed your previous post on the level shifters -- just read my emails and saw it and came back here to update..  I also just read the post from bigdinotech  on the update to the i2c lib for reducing the delays in the I2C lib. Will give that a try first to see what I get but probably won't happen until sometime this weekend.

mjs513, 2016-06-25:
Ok. I couldn't sleep so I decided to go ahead and test the timing changes for the 6050:

I2C_SPEED_FAST: 0.77 ms 
I2C_SPEED_SLOW: 0.87 ms

It is definitely better. I will check the mega again tomorrow.

piroozn, 2016-06-25:
Thanks Larry,

Pirooz

Thanks for your help. There does seem to be alot of questions out there with the I2C bus on the 101 board. By the way if you want one of the sensors send me an address and I can send it to you (CyberMerln@gmail.com<mailto:CyberMerln@gmail.com<mailto:CyberMerln@gmail.com%3cmailto:CyberMerln@gmail.com>>).

mjs513, 2016-06-25:
@PaulStoffregen I agree with you 100% on performance.  I have been using and constantly updating Fabio Varesano FreeIMU code for the last couple of years and came across the I2C issues while I was testing both the FreeIMU library and adding a Magnetometer to the CurieIMU (bmi160).  The performance was very poor on the Arduino101 I started investigating the read and performance speeds using Fabio's original sketch and modifying to just focus on read speeds.  For the heck of it I attached the PropShield that I picked up for use with the Teensy 3.2 and really saw the performance degradation for both the 6050 and the PropShield.  Plan now is when I get some time to load FreeIMU and document the performance.  I still would have thought that the read speeds would be comparable to the Mega at 400khz which will affect performance.

mjs513, 2016-06-25:
Would still recommend  the following change to the wire.cpp file to add the following function to make a similar call to a wire.setClock function:

```
void TwoWire::setClock(long speed)
{
	if(speed == 400000L) {
		init_status = i2c_openadapter_speed(I2C_SPEED_FAST);
	} else if(speed == 100000L) {
		init_status = i2c_openadapter_speed(I2C_SPEED_SLOW);
	} else {
		init_status = i2c_openadapter();
	}
}
```
Your call.

mjs513, 2016-06-26:
ALCON.  I ran another test of the 6050 on the Mega and 101 boards using the FreeIMU library and rerunning the I2C speed test again. Here are the results:

FreeIMU Performance Tests:

 Mega 2560:
> Testing raw reading speed (average on 512 samples):
> --> result: 356 microseconds .... 0 milliseconds
> Testing calibrated reading speed (average on 512 samples):
> --> result: 742 microseconds .... 0 milliseconds
> Testing sensor fusion speed (average on 512 samples):
> --> result: 2753 microseconds .... 2 milliseconds
> Looping again..
> ----

Arduino 101 (i2cFast):

> Testing raw reading speed (average on 1024 samples):
> --> result: 34905 microseconds .... 34 milliseconds
> Testing calibrated reading speed (average on 1024 samples):
> --> result: 34920 microseconds .... 34 milliseconds
> Testing sensor fusion speed (average on 1024 samples):
> --> result: 35026 microseconds .... 35 milliseconds
> Looping again..

A little surprised so I reran the i2C speed test for sensor reads on the Arduino 101:

> Testing raw reading speed (average on 1024 samples):
> --> result: 16851 microseconds .... 16.85186 milliseconds
> Looping again..

So I am not sure now what is really going on with timing results unless I was seeing hallucinating the first time through.  I am using 1.6.9 but got the same results with using 1.6.7.  I checked these results about 10 times since last night.  I am now totally lost in why the differences from load to load.  If you all have a different way to test read speeds post the code and I will give it a try.

Mike

piroozn, 2016-06-26:
Hi @mjs513
Looks like I was looking at the voltage translator wrong. You don’t even need the internal pullup.
If you want to measure the timing using another method then you can toggle a pin and measure it with scope. Before and after the operation to measure.
Can you change your start and stop micros() in you code to measure the wire.write statement to measure the I2C write or read and measure with fast or slow clock or even comparing it with mega.
Looks like this is a code issue and not the speed of the I2C.

mjs513, 2016-06-26:
Hi @piroozn I would say it is a code issue but the same code is used on the mega and 101, only difference is the libraries and board code.  Yes I can do timing but I will have to get modify the library to do the measuring.  I can try something else but it will take a little time to get the modifications done.

piroozn, 2016-06-26:
Hi @mjs513
The following is the test result comparing Arduino 101 with Uno board using RGB-Display Sketch.
Looks like some one has added a delay after every display function. I remoed these delays and now you can see the delta in timing when you change clock speed. You can also see that Arduino 101 is faster than Uno board. Removing these delays did not create any issue as fasr as displaying goes.

Arduino 101 test result

default
--> result: 2800286 microseconds .... 
--> result: 2800271 microseconds .... 
--> result: 2800267 microseconds .... 
--> result: 2800275 microseconds .... 
--> result: 2800264 microseconds .... 
--> result: 2800276 microseconds .... 
--> result: 2800267 microseconds .... 
--> result: 2800277 microseconds .... 
--> result: 2800269 microseconds .... 
--> result: 2800271 microseconds ....  
Test 2
--> result: 5311 microseconds .... 
--> result: 5312 microseconds .... 
--> result: 5312 microseconds .... 
--> result: 5312 microseconds .... 
--> result: 5312 microseconds .... 
--> result: 5312 microseconds .... 
--> result: 5311 microseconds .... 
--> result: 5312 microseconds .... 
--> result: 5311 microseconds .... 
--> result: 5313 microseconds .... 
--> result: 5313 microseconds .... 
--> result: 5313 microseconds .... 
--> result: 5313 microseconds .... 
--> result: 5313 microseconds .... 
--> result: 5314 microseconds ....

FAST
--> result: 1059632 microseconds .... 
--> result: 1059621 microseconds .... 
--> result: 1059614 microseconds .... 
--> result: 1059619 microseconds .... 
--> result: 1059616 microseconds .... 
--> result: 1059619 microseconds .... 
--> result: 1059613 microseconds .... 
--> result: 1059616 microseconds .... 
--> result: 1059618 microseconds .... 
--> result: 1059618 microseconds .... 
--> result: 1059619 microseconds .... 
--> result: 1059616 microseconds .... 
--> result: 1059618 microseconds .... 
--> result: 1059622 microseconds .... 
--> result: 1059614 microseconds .... 
--> result: 1059616 microseconds .... 
--> result: 1059625 microseconds ...

Test 2
--> result: 2033 microseconds .... 
--> result: 2033 microseconds .... 
--> result: 2032 microseconds .... 
--> result: 2032 microseconds .... 
--> result: 2032 microseconds .... 
--> result: 2033 microseconds .... 
--> result: 2033 microseconds .... 
--> result: 2032 microseconds .... 
--> result: 2033 microseconds .... 
--> result: 2032 microseconds .... 
--> result: 2033 microseconds .... 
--> result: 2032 microseconds .... 
--> result: 2033 microseconds .... 
--> result: 2032 microseconds ....  

Slow
--> result: 2800282 microseconds .... 
--> result: 2800270 microseconds .... 
--> result: 2800264 microseconds .... 
--> result: 2800271 microseconds .... 
--> result: 2800263 microseconds .... 
--> result: 2800276 microseconds .... 
--> result: 2800266 microseconds .... 
--> result: 2800272 microseconds .... 
--> result: 2800275 microseconds .... 
--> result: 2800269 microseconds .... 
--> result: 2800273 microseconds ....  

Test2
--> result: 260854 microseconds .... 
--> result: 260846 microseconds .... 
--> result: 260846 microseconds .... 
--> result: 260845 microseconds .... 
--> result: 260844 microseconds .... 
--> result: 260845 microseconds .... 
--> result: 260845 microseconds .... 
--> result: 260845 microseconds .... 
--> result: 260845 microseconds .... 
--> result: 260844 microseconds .... 
--> result: 260845 microseconds .... 
--> result: 260844 microseconds .... 
--> result: 260844 microseconds .... 
--> result: 260845 microseconds .... 
--> result: 260845 microseconds ....  

--------------------------------------------------------

Arduino Uno test result
Test1
--> result: 2858024 microseconds .... 
--> result: 2858116 microseconds .... 
--> result: 2858084 microseconds .... 
--> result: 2858032 microseconds .... 
--> result: 2858100 microseconds .... 
--> result: 2858076 microseconds .... 
--> result: 2858092 microseconds .... 
--> result: 2858100 microseconds .... 
--> result: 2858028 microseconds .... 
--> result: 2858132 microseconds .... 
--> result: 2858076 microseconds .... 
--> result: 2858092 microseconds .... 
--> result: 2858100 microseconds .... 
--> result: 2858028 microseconds ....  

Test2
--> result: 5164 microseconds .... 
--> result: 5164 microseconds .... 
--> result: 5184 microseconds .... 
--> result: 5176 microseconds .... 
--> result: 5172 microseconds .... 
--> result: 5164 microseconds .... 
--> result: 5168 microseconds .... 
--> result: 5180 microseconds .... 
--> result: 5188 microseconds .... 
--> result: 5168 microseconds .... 
--> result: 5164 microseconds .... 
--> result: 5164 microseconds .... 
--> result: 5180 microseconds .... 
--> result: 5188 microseconds .... 
--> result: 5172 microseconds ....

piroozn, 2016-06-26:
Test 2

/*
 * Example for Shield Arduino FRDM STBC AGM01 of Freescale
 * Author: Andres Sabas 
 * 17 Agust 2015
 * 
 * 
 */
//#include <Wire.h>
#include "Wire.h"

//#include <rgb_lcd.h>
#include "rgb_lcd.h"

rgb_lcd lcd;

extern unsigned long start, stop, speed;

const int colorR = 50;
const int colorG = 0;
const int colorB = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  
  // Initialize the FXOS8700CQ
  //sensor.init();
  
  // Initialize the FXAS21002C
//  sensor2.init();

      // set up the LCD's number of columns and rows:
    lcd.begin(16, 2);

    lcd.setRGB(colorR, colorG, colorB);
    
    // Print a message to the LCD.
    lcd.print("hello, world!");

    delay(1000);
}
void loop() {
  // put your main code here, to run repeatedly:

  speed = 0;
  start = micros();

    // set the cursor to column 0, line 1
    // (note: line 1 is the second row, since counting begins with 0):
    lcd.setCursor(0, 1);
    // print the number of seconds since reset:
    lcd.print(millis()/1000);

    

  
    lcd.print("hello, world!");
 /* 
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
[... 39 lines cut ...]
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
 lcd.print("Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing Testin Testing Testing Testing Testing Testing Testing Testing Testing Testing");
*/

 stop = micros();
  speed = stop - start;
 Serial.print("--> result: ");
 Serial.print((speed));
 //Serial.print((stop - start));
 //Serial.print((stop - start) / 1024);
 //Serial.print(" microseconds .... ");
 Serial.println(" microseconds .... ");
 //Serial.print(((stop - start) / 1024) / 1000);
 //Serial.println(" milliseconds"); 
  
}
@mjs513

piroozn, 2016-06-26:
@mjs513
Test 1 displays lots of lines
Test2 display 1 line and time passed in second so I can see if the display is correct or not.

mjs513, 2016-06-27:
@piroozn So basically you had to modify library to remove the delays. Ok I went the code in I2Cdev lib and the mpu6050 lib just to make sure I did not have an issue with WIre.begin and no wire.begin's were found in the libs.  In the MPU6050 lib the single function that gets called is getMotion6:
```
void MPU60X0::getMotion6(int16_t* ax, int16_t* ay, int16_t* az, int16_t* gx, int16_t* gy, int16_t* gz) {
   I2Cdev::readBytes(bSPI, devAddr, MPU60X0_RA_ACCEL_XOUT_H, 14, buffer);
    *ax = (((int16_t)buffer[0]) << 8) | buffer[1];
    *ay = (((int16_t)buffer[2]) << 8) | buffer[3];
    *az = (((int16_t)buffer[4]) << 8) | buffer[5];
    *gx = (((int16_t)buffer[8]) << 8) | buffer[9];
    *gy = (((int16_t)buffer[10]) << 8) | buffer[11];
    *gz = (((int16_t)buffer[12]) << 8) | buffer[13];
}
```
the only function in I2Cdevlib that gets called from the sketch is as follows:

```
/** Read multiple bytes from an 8-bit device register.
 * @param useSPI  true : use SPI 
 * @param devAddr I2C slave device address
 * @param regAddr First register regAddr to read from
 * @param length Number of bytes to read
 * @param data Buffer to store read data in
 * @param timeout Optional read timeout in milliseconds (0 to disable, leave off to use default class value in I2Cdev::readTimeout)
 * @return Number of bytes read (0 indicates failure)
 */
int8_t I2Cdev::readBytes(bool useSPI, uint8_t devAddr, uint8_t regAddr, uint8_t length, uint8_t *data, uint16_t timeout) {
    #ifdef I2CDEV_SERIAL_DEBUG
        Serial.print(useSPI ? "SPI (0x" : "I2C 0x");
        Serial.print(devAddr, HEX);
        Serial.print(") reading ");
        Serial.print(length, DEC);
        Serial.print(" bytes from 0x");
        Serial.print(regAddr, HEX);
        Serial.print("...");
    #endif
    int8_t count = 0;
	// I2C
	if (!useSPI) {
		Wire.beginTransmission(devAddr);
		#if ((I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO < 100) || I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_NBWIRE)
			Wire.send(regAddr);
		#elif (I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO >= 100)
			Wire.write(regAddr);
		#endif
		Wire.endTransmission();
		Wire.beginTransmission(devAddr);
		Wire.requestFrom(devAddr, length);

		uint32_t t1 = millis();
		for (; Wire.available() && (timeout == 0 || millis() - t1 < timeout); count++) {
			#if ((I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO < 100) || I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_NBWIRE)
				data[count] = Wire.receive();
			#elif (I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO >= 100)
				data[count] = Wire.read();
			#endif
			#ifdef I2CDEV_SERIAL_DEBUG
				Serial.print(data[count], HEX);
				if (count + 1 < length) Serial.print(" ");
			#endif
		}
		if (timeout > 0 && millis() - t1 >= timeout && count < length) count = -1; // timeout
		Wire.endTransmission();
	} else {
	    digitalWrite(devAddr, LOW);
		byte Addr = regAddr | 0x80;
		SPI.transfer(Addr);
		for (uint8_t cnt=0; cnt < length; cnt++) {
			data[cnt] = SPI.transfer(0);
			count++;
		}
		digitalWrite(devAddr, HIGH);
	}
    #ifdef I2CDEV_SERIAL_DEBUG
        Serial.print(". Done (");
        Serial.print(count, DEC);
        Serial.println(" read).");
    #endif
    return count;
}
```
As you can see there are no delays in the read register functions and there is no delays in the sketch itself.  Only delays I found were during IMU initialization.  THe only potetial problem is with the ARDUINO >= 100 statement - is this still valid for the Arduino101 board?

There are a couple of things that I want to verify because it seems like something is kicking it back to the official core release. The other thing is where do you recommend be putting the timing test in the getMotion6 function?

Thanks Mike

mjs513, 2016-06-27:
Test 1 - timing loop on wire.read as follows:
```
			#elif (I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO >= 100)
				long start = micros();
				data[count] = Wire.read();
				Serial.println(micros() - start);
			#endif
```
This yields a value of 1 or 2 microseconds per read for both fast or slow i2c setting.

Test 2 timing:
```
		long start = micros();
		for (; Wire.available() && (timeout == 0 || millis() - t1 < timeout); count++) {
			#if ((I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO < 100) || I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_NBWIRE)
				data[count] = Wire.receive();
			#elif (I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO >= 100)
				data[count] = Wire.read();
			#endif
			#ifdef I2CDEV_SERIAL_DEBUG
				Serial.print(data[count], HEX);
				if (count + 1 < length) Serial.print(" ");
			#endif
		}
		Serial.println(micros() - start);
		if (timeout > 0 && millis() - t1 >= timeout && count < length) count = -1; // timeout
		Wire.endTransmission();
```
This yield approximately 118 microseconds (14 byte read) vs 1 byte reads in test1 for both fast and slow.

TEST 3 
```
	long start = micros();
    #ifdef I2CDEV_SERIAL_DEBUG
        Serial.print(useSPI ? "SPI (0x" : "I2C 0x");
        Serial.print(devAddr, HEX);
        Serial.print(") reading ");
        Serial.print(length, DEC);
        Serial.print(" bytes from 0x");
        Serial.print(regAddr, HEX);
        Serial.print("...");
    #endif
    int8_t count = 0;
	// I2C
	if (!useSPI) {
		Wire.beginTransmission(devAddr);
		#if ((I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO < 100) || I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_NBWIRE)
			Wire.send(regAddr);
		#elif (I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO >= 100)
			Wire.write(regAddr);
		#endif
		Wire.endTransmission();
		Wire.beginTransmission(devAddr);
		Wire.requestFrom(devAddr, length);

		uint32_t t1 = millis();
		for (; Wire.available() && (timeout == 0 || millis() - t1 < timeout); count++) {
			#if ((I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO < 100) || I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_NBWIRE)
				data[count] = Wire.receive();
			#elif (I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE && ARDUINO >= 100)
				data[count] = Wire.read();
			#endif
			#ifdef I2CDEV_SERIAL_DEBUG
				Serial.print(data[count], HEX);
				if (count + 1 < length) Serial.print(" ");
			#endif
		}
		if (timeout > 0 && millis() - t1 >= timeout && count < length) count = -1; // timeout
		Wire.endTransmission();

	} else {
	    digitalWrite(devAddr, LOW);
		byte Addr = regAddr | 0x80;
		SPI.transfer(Addr);
		for (uint8_t cnt=0; cnt < length; cnt++) {
			data[cnt] = SPI.transfer(0);
			count++;
		}
		digitalWrite(devAddr, HIGH);
	}
    #ifdef I2CDEV_SERIAL_DEBUG
        Serial.print(". Done (");
        Serial.print(count, DEC);
        Serial.println(" read).");
    #endif
	Serial.println(micros() - start);

``````
16890 micros for fast and 18056 micros in slow mode

I will repeat these same tests on the mega2560 when I get a chance.

mjs513, 2016-06-27:
Ok.  I repeated test 3 with the Mega, here are the results
----------------------------------------------------------------------------------
100khz: 1088 micros.  There are spikes to 17000 periodically.
400khz: 175 micros

Test 2:
400khz: 5 micros.
100khz: 5 micros
---------------------------------------------------------------------------
So the driver is to I2C timing for me is in that one function call to readBytes.

piroozn, 2016-06-27:
@mjs513
When I was testing the RGB-Display, notices timing was different when using the wire.cpp and wire.h from the main library. I copied the source in my local directory and repeated the test and it worked ok. The library file .a may not be up to date with the new changes and that may be the reason for your test result. Do you have wire.cpp in your local directory?
I think you also need to copy the source for the lib files for your sensors (FXOS8700CQ, FXAS21002C) since they may reset the I2C clock back to default. Keep then in your local directory and check for wire.begin(????).

bigdinotech, 2016-06-27:
@mjs513 did your latest tests include the fixes from
https://github.com/bigdinotech/corelibs-arduino101/tree/i2cdelay

Also if that is still too slow you can try reducing the delays in i2c.c 
JUst make sure the total timeout value remains the same

I will add in your recommended changes after I test it out on our side.

mjs513, 2016-06-27:
@piroozn All my tests were done using a GY-521 MPU6050 breakout board.  The libraries used were one for the MPU6050 and a version of the I2Cdev lib by jeff rowberg.  Both libraries I have checked a few times and no wire.begin calls in either library.  the I2Cdev lib uses the standard wire library calls.  I have no wire.cpp files in my local directory.  I use the standard libs associated with the board.
If the issue is putting library source files in the local directory kind of defeats the purpose of having the libraries.

@bigdinotech Yes I did install the i2cdelay version.  Playing on reinstalling just in case something changed it.  I looked in the i2c.cpp file and there are a few delayMicroseconds(50) calls.  So I am not sure what you mean about total timeout the same.

Thanks
Mike

bigdinotech, 2016-06-27:
static int wait_tx_or_err(bool no_stop){
	uint64_t timeout = TIMEOUT_MS * 20;
	while(timeout--) {
		if (i2c_err_detect) {
			return I2C_ERROR;
		}
		if (!no_stop) {
			if (i2c_tx_complete) {
			return I2C_OK;
			}
		}
		delayMicroseconds(50);
	}
	if (!no_stop)
		return I2C_TIMEOUT;
	else
		return I2C_OK;
}

the delayMicroseconds(50) is inside a while loop.
so if you decrease the delay you will have to increase the timeout value.
so if you reduce the delay to delayMicroseconds(10)
then you will have to change it to uint64_t timeout = TIMEOUT_MS * 100;

mjs513, 2016-06-27:
@bigdinotech Thanks. Will play around with it in the next few days and let you know what I find out.

piroozn, 2016-06-27:
@bigdinotech
Do you know why we are using these delay after every I2C access. The I2C controller has all sort of status capability and we should not be needing any delay. We should only use time out and that will only trigger if an error occur and it should be only once in a while if any.
Any new I2C write or read should check status and if FIFO is full or empty then an action may be needed depending to the operation.

mjs513, 2016-06-28:
@bigdinotech Ok I played around with delays and it seemed to make no  difference between my previous results if I follow your instructions.  If I follow what the user on the forum suggested and just set uint64_t timeout = TIMEOUT_MS; the numbers are 1.588 millis (slow) and 1.411millis (fast) almost no difference between fast and slow but big diff in the timingins for the reads.

piroozn, 2016-06-28:
@mjs513
Did you copy the wire.cpp to your local directory? I know that is not norm but Dino just made this change and it may not be part of the .a file. I think it did not work for me either until I copies the code in my working directory.
The fact that timeout is effecting your timing may be that you have so much delay that timeout is kicking in every time to make any differences. Time out is supposed to only get hit in case of an error.
The followings are the fact since I used RGB-Display board instead of your hardware configuration to prove the I2C speed changes.
-	The clock speed change had large impact on the timing of the I2C operation
-	Arduino 101 outperformed Ardunio Uno for the I2C writes.
-	The wire.cpp and wire.h source had to be copied to the working directory and include filed in the RGB driver have to be changed to point to it “wire.cpp” instead of <wire.cpp>
-	Removed all delay statement after any I2C access in the rgb_lcd.cpp drivers also located in the working directory.
Assumption:
-	Your IMU performance related to the I2C timing is cause by the same delay statement after every I2C access

mjs513, 2016-06-28:
Ok.
The wire.cpp and wire.h source had to be copied to the working directory : copied
The wire.cpp and wire.h source had to be copied to the working directory and include filed in the RGB driver have to be changed to point to it “wire.cpp” instead of : done - only library that uses wire is for the MPU6050 library.  Assumed working directory is the sketch directory.

Here are the errors it threw:

> C:\Users\CyberPalin\Documents\Arduino\libraries\I2Cdev/I2Cdev.h:70:28: fatal error: Wire.cpp: No such file or directory
> 
>          #include <Wire.cpp>
>                             ^
> compilation terminated.

mjs513, 2016-06-28:
Hi @piroozn 
Ok - changed the sketch that I am using completely to avoid using any external libraries.  Thanks to Kris Winer's 6050 code (on GitHub) I put timing statements around readAccel data and readGyro data which as well as around the actual wire.read statements. I also put the wire.h and wire.cpp in the sketch folder.  Here are the results:

> 
> slow mode
> Read time for gyro and accel data: 33901
> Read time for gyro and accel data: 33899
> Read time for gyro and accel data: 33901
> Read time for gyro and accel data: 33900
> Read time for gyro and accel data: 33900
> Read time for gyro and accel data: 33900
> Read time for gyro and accel data: 33900
> Read time for gyro and accel data: 33899
> Read time for gyro and accel data: 33899
> Read time for gyro and accel data: 33900
> X-acceleration: 3.17 mg Y-acceleration: -995.61 mg Z-acceleration: 87.65 mg
> X-gyro rate: 1.0 degrees/sec Y-gyro rate: -0.3 degrees/sec Z-gyro rate: 0.3 degrees/sec
> Temperature is 41.99 degrees C
> 
> fast mode
> Read time for gyro and accel data: 32900
> Read time for gyro and accel data: 32901
> Read time for gyro and accel data: 32899
> Read time for gyro and accel data: 32899
> Read time for gyro and accel data: 32900
> Read time for gyro and accel data: 32897
> Read time for gyro and accel data: 32899
> Read time for gyro and accel data: 32901
> Read time for gyro and accel data: 32901
> Read time for gyro and accel data: 32900
> X-acceleration: 1.95 mg Y-acceleration: -996.09 mg Z-acceleration: 90.82 mg
> X-gyro rate: 0.6 degrees/sec Y-gyro rate: -0.6 degrees/sec Z-gyro rate: 0.4 degrees/sec
> Temperature is 42.13 degrees C

Not much difference between speeds.  Second as a note these are higher because they are doing two read data calls while the lib version got the necessary data in on shot.

Here is the code:
/* MPU6050 Basic Example Code
 by: Kris Winer
 date: May 1, 2014
 license: Beerware - Use this code however you'd like. If you 
 find it useful you can buy me a beer some time.
 
 Demonstrate  MPU-6050 basic functionality including initialization, accelerometer trimming, sleep mode functionality as well as
 parameterizing the register addresses. Added display functions to allow display to on breadboard monitor. 
 No DMP use. We just want to get out the accelerations, temperature, and gyro readings.
 
 SDA and SCL should have external pull-up resistors (to 3.3V).
 10k resistors worked for me. They should be on the breakout
 board.
 
 Hardware setup:
 MPU6050 Breakout --------- Arduino
 3.3V --------------------- 3.3V
 SDA ----------------------- A4
 SCL ----------------------- A5
 GND ---------------------- GND
 
  Note: The MPU6050 is an I2C sensor and uses the Arduino Wire library. 
 Because the sensor is not 5V tolerant, we are using a 3.3 V 8 MHz Pro Mini or a 3.3 V Teensy 3.1.
 We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.
 We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ  to 400000L /twi.h utility file.
 */
 
#include "Wire.h"

// Define registers per MPU6050, Register Map and Descriptions, Rev 4.2, 08/19/2013 6 DOF Motion sensor fusion device
// Invensense Inc., www.invensense.com
// See also MPU-6050 Register Map and Descriptions, Revision 4.0, RM-MPU-6050A-00, 9/12/2012 for registers not listed in 
// above document; the MPU6050 and MPU-9150 are virtually identical but the latter has an on-board magnetic sensor
[... 622 lines cut ...]
  Wire.write(data);                 // Put data in Tx buffer
  Wire.endTransmission();           // Send the Tx buffer
}

  uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data; // `data` will store the register data   
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
  Wire.requestFrom(address, (uint8_t) 1);  // Read one byte from slave register address 
  data = Wire.read();                      // Fill Rx buffer with result
  return data;                             // Return data read from slave register
}

  void readBytes(uint8_t address, uint8_t subAddress, uint8_t count, uint8_t * dest)
{  
  Wire.beginTransmission(address);   // Initialize the Tx buffer
  Wire.write(subAddress);            // Put slave register address in Tx buffer
  Wire.endTransmission(false);       // Send the Tx buffer, but send a restart to keep connection alive
  uint8_t i = 0;
        Wire.requestFrom(address, count);  // Read bytes from slave register address 
  while (Wire.available()) {
        dest[i++] = Wire.read(); }         // Put read results in the Rx buffer
}

bigdinotech, 2016-06-28:
Ok. I did some more changes. And all so ran some tests on my side.

I ran a test on using an I2C FRAM breakout board.
My test simply runs a tight loop of a write followed by a read.

[image]

As you can see in the attached image the max delay I am getting between read and write operations is about 25 us.

Changes/Amends I made can be found here:
https://github.com/bigdinotech/corelibs-arduino101/tree/i2cdelay

piroozn, 2016-06-28:
@bigdinotech
Looks like the 25us gap is for other interrupt or tasks and they are not all the same. The data transmission sizes are all different also. If you zoom in then we can see the delay between each read or write access. These gaps are not the delay and is most likely interrupts or other tasks taking over.
What is the test code looks like

mjs513, 2016-06-28:
@bigdinotech 
I see the changes you made to the delays,  No sure what other changes you made.  But I just ran the sketch with the changes and they got significantly worse:

```
Read time for gyro and accel data: 44527

Read time for gyro and accel data: 44516

Read time for gyro and accel data: 44525

Read time for gyro and accel data: 44515

Read time for gyro and accel data: 44515

Read time for gyro and accel data: 44516

Read time for gyro and accel data: 44525

Read time for gyro and accel data: 44514

Read time for gyro and accel data: 44523

Read time for gyro and accel data: 44527

Read time for gyro and accel data: 44525
X-acceleration: 1008.54 mg Y-acceleration: 50.78 mg Z-acceleration: 31.25 mg
X-gyro rate: 126.1 degrees/sec Y-gyro rate: 6.3 degrees/sec Z-gyro rate: 3.9 degrees/sec
Temperature is 127.59 degrees C

```

mjs513, 2016-06-28:
@bigdinotech 
Ok I did some further testing with the versions of the sketches that I have used for the 6050.  Here are the results.  Interesting that the original sketch improved a lot:

> Kris Ws Sketch with wire in folder
> Wire.begin(I2C_SPEED_FAST)
> Read time for gyro and accel data: 39510
> Read time for gyro and accel data: 39507
> Read time for gyro and accel data: 39508
> Read time for gyro and accel data: 39504
> Read time for gyro and accel data: 39507
> Read time for gyro and accel data: 39508
> Read time for gyro and accel data: 39507
> Read time for gyro and accel data: 39507
> Read time for gyro and accel data: 39506
> Read time for gyro and accel data: 39508
> Read time for gyro and accel data: 39506
> 
> Wire.begin(I2C_SPEED_SLOW)
> Read time for gyro and accel data: 39571
> Read time for gyro and accel data: 39570
> Read time for gyro and accel data: 39572
> Read time for gyro and accel data: 39571
> Read time for gyro and accel data: 39570
> Read time for gyro and accel data: 39570
> Read time for gyro and accel data: 39569
> Read time for gyro and accel data: 39570
> Read time for gyro and accel data: 39569
> Read time for gyro and accel data: 39570
> Read time for gyro and accel data: 39571
> 
> setClock(400000L)
> no diff
> 
> setClock(100000L)
> no diff
> 
> ------------------------------
> Library Sketch (original test
> Wire.begin(I2C_SPEED_SLOW); --> result: 6267 microseconds .... 6.26783 milliseconds
> 
> Wire.begin(I2C_SPEED_FAST); --> result: 6096 microseconds .... 6.09659 milliseconds
> 
> Wire.setClock(100000L); --> result: 5270 microseconds .... 5.27056 milliseconds
> 
> Wire.setClock(400000L); --> result: 5094 microseconds .... 5.09442 milliseconds

The one question i have is why is keeping timeout x delay has to be set to 1000?

mjs513, 2016-06-29:
@bigdinotech Ok Dino. I broke down and bought a real oscilloscope - should be here next week.  Anyway I came across a couple of things: https://communities.intel.com/thread/56148 and something from intel on interfacing i2c and the smbus: http://www.intel.com/content/www/us/en/embedded/technology/connectivity/smbus-controller-i2c-devices-paper.html.  Haven't gone through it yet but it looks interesting.

PaulStoffregen, 2016-07-01:
Does the Arduino 101 hardware have the capability to run at 1 MHz I2C clock rate?

I just ordered some MB85RC256 FRAM chips for testing.  This chip supports 100 kHz, 400 kHz and 1 MHz rates, and can run from 2.7V to 5.5V range, so it ought to cover testing all the various Arduino boards nicely.

In the meantime, I'll try to do some initial tests with EEPROM chips I have on hand, but which only work at 100 & 400 kHz.

mjs513, 2016-07-01:
HI Paul

Where did you order the eeprom from?

PaulStoffregen, 2016-07-01:
I'm testing right now with a 24C01A EEPROM which is an old part I just happen to have laying around.

[image]

The FRAM chip I ordered just now from Digikey.

http://www.digikey.com/product-detail/en/fujitsu-electronics-america-inc/MB85RC256VPNF-G-JNERE1/865-1261-1-ND

PaulStoffregen, 2016-07-01:
Ok, I've run the test on 5 different boards.  I tested Arduino 101 twice, with the 1.0.6 core and with bigdinotech's i2cdelay branch.  Here's the time in microseconds required to read 12 bytes.

    Board Tested                100 kHz         400 kHz 
    ------------                -------         ------- 
    Arduino Uno                 1540            512
    Arduino Due                 1404            366
    Arduino Zero                1469            448     
    Teensy 3.2                  1404            390
    Arduino 101 (1.0.6)         18104
    Arduino 101 (bigdinotech)   35509           34625

Here is the code I used for testing:

    #include <Wire.h>

    #define SPEED  100000L
    //#define SPEED  400000L
    //#define SPEED  1000000L

    #define WIRE Wire   // Uno, 101, et
    //#define WIRE Wire1    // Due

    void setup() {
      Serial.begin(9600);
      while (!Serial) ; // wait for serial monitor
      Serial.println("EEPROM Read Test");
      WIRE.begin();
      WIRE.setClock(SPEED);
      delay(100);
    }

    void loop() {
      unsigned char buf[16];
      int i, n=0;
      unsigned long tbegin, tend;

      tbegin = micros();
      WIRE.beginTransmission(0x50);
      WIRE.write(0); // address within chip
      WIRE.endTransmission(false);
      WIRE.requestFrom(0x50, 12);
      while (WIRE.available() && n < 12) {
        buf[n++] = WIRE.read();
      }
      tend = micros();

      Serial.println();
      Serial.print("Read ");
      Serial.print(n);
      Serial.print(" bytes in ");
      Serial.print(tend - tbegin);
      Serial.println(" us");

      Serial.print("Data as integers:  ");
      for (i=0; i < n; i++) {
        Serial.print(" ");
        Serial.print((unsigned int)buf[i]);
      }
      Serial.println();

      Serial.print("Data as characters:");
      for (i=0; i < n; i++) {
        Serial.print(" ");
        Serial.print((char)buf[i]);
      }
      Serial.println();

      delay(1000);
    }

PaulStoffregen, 2016-07-01:
Here's some oscilloscope captures for the 400 kHz test with bigdinotech i2cdelay branch.

First, here you can see the whole 34625 time.  Arduino 101 seems to be working, but there's some sort of terrible bug which just stalls for a very long time in the middle of the transfer. 

[image]

Here's a zoom in to the first part:

[image]

and here's a zoom in to the large part

[image]

so the good news is I2C seems to be working, except for some horrible bug in 101's Wire library that's just stalling for a very long time.

As a final check, here's a zoom in to a small section.  Looks like 101 is actually using 444 kHz clock rate.  A little over the spec, but not too bad.

[image]

Hopefully this testing and these screenshots help a bit?

mjs513, 2016-07-01:
Just ordered one from adafruit. Thanks for testing and code as well. At least I know i am not going crazy.

PaulStoffregen, 2016-07-01:
Yup, you're definitely not going crazy.  There's clearly a very bad bug in Arduino 101's Wire library.

mjs513, 2016-07-01:
Looks you were right with your first comment on the i2c is only as good as the wire library. Thank you.

PaulStoffregen, 2016-07-01:
I you want to try this test yourself on a 24C01A chip, here's the code I used to write initial data.

    #include <Wire.h>

    void setup() {
      Serial.begin(9600);
      while (!Serial) ; // wait for serial monitor
      Serial.println("EEPROM Write Test");
      Wire.begin();
      Wire.setClock(100000L);
      delay(100);
      Wire.beginTransmission(0x50);
      Wire.write(0); // address within chip
      Wire.print("Hello Wo");
      Wire.endTransmission();
      Serial.println("Wrote first 8 bytes");
      delay(100);
      Wire.beginTransmission(0x50);
      Wire.write(8); // address within chip
      Wire.println("rld!");
      Wire.endTransmission();
      Serial.println("Wrote the rest");
    }

    void loop() {
    }

The test should run without this step, but it's much more interesting and reassuring to see the EEPROM read recognizable data.  And if you write the chip with this code, you should end up with the exact same waveforms.

mjs513, 2016-07-01:
I will definitely try it. Will probably get the Breakout board and the new osciliscope about the same time. So now I will have a couple new toys.

[33 more replies]

closed 2016-07-09 by mjs513