arduino/ArduinoCore-arc32 #168: Can't upgrade my Aruduino 101 board version on Arduino 1.6.8

dgdosen, 2016-04-16:
I just picked up an Arduino 101 and tried to run a sketch a bit more complicated than blink...

First off - I get an error trying to load Board Support via the Arduino Board Manager to 1.0.5 because of this error:

```
Error downloading https://github.com/01org/corelibs-arduino101/archive/1.6.4+1.68.zip
```

That's pointing to this repo.... and to a file that does not exist.  Any ideas on how to fix this?

bbaltz505, 2016-04-16:
Hi.
Is this still an issue for you? I had another report of the download not working. GitHub may have been down briefly.
Brian

dgdosen, 2016-04-19:
Seems to be resolved, but I don't think it was caused by github being down.  I think it was caused by the link not working...

closed 2016-04-19 by dgdosen