arduino/ArduinoCore-arc32 #175: Cannot pull versions 1.0.5 or 1.0.4 from the Arduino Boards Manager

Halfloafwhr, 2016-05-06:
Hello! I received my first Arduino101 in the mail yesterday, and when trying to add the board via the Boards Manager, I see the error "Error downloading https://github.com/01org/corelibs-arduino101/archive/1.6.4+1.68.zip". 

Is this an error with how the code is structured here at Github, or is it an error with the information stored in the boards manager?

Thank you!

Halfloafwhr, 2016-05-06:
Well, I just clicked the link in my comment above, and received a .zip file.  The installation still will not function, though.  How do I manually associate this with the board in the Arduino IDE?

eriknyquist, 2016-05-06:
Hi @Halfloafwhr, could you give details about your operating system please?

calvinatintel, 2016-05-09:
Have you tried following the guide on Readme? https://github.com/01org/corelibs-arduino101/blob/master/README.md

bbaltz505, 2016-05-09:
Another user (issue #168) reported a link failure during a Board Manager install.He retried shortly later and it worked. Links to releases seem to fail at times it seems (both on Fridays in this case). 

The easiest solution would be to uninstall the Curie package (if it is listed in Board Manager) and reinstall.

Halfloafwhr, 2016-05-11:
Hi!  My operating system is Windows 7, 64-bit.  I have tried the guide on the ReadMe, but after copying the files over to the proper directory, I cannot select the Arduino101 on the list of installed boards.

Halfloafwhr, 2016-05-11:
Ah! It looks like it was an issue with my company's firewall.  It has been resolved.  For some reason, I couldn't download the drivers for Intel chips, while I could for the other arduino boards that I'm using.  

Thank you for the time and attention!

closed 2016-05-11 by Halfloafwhr