arduino/ArduinoCore-arc32 #177: _value buffer allocation error in CurieBLE BLECharacteristic.cpp

twobones, 2016-05-06:
On line 40 of BLECharacteristic.cpp, the code in the constructor is "malloc(_value_length)".  However _value_length was initialized to 0.  I believe the code should read "malloc(_value_size)" which was just calculated.  Testing indicates that malloc(0) seems to round up to 16 bytes of allocation, because any BLE message of 16 bytes or less works fine.  Messages longer than this overflow the buffer into whatever was allocated next from the heap.

SidLeung, 2016-05-06:
This is a bug.  A JIRA ticket has been created.  Will be corrected on the next Core Library (JSON) release.  Also, please note that there is a #define limit on the buffer size.  You can increase that limition to less than or equal to 200 bytes, if needed.

SidLeung, 2016-06-01:
This issue is corrected in the Bellatrix release.

closed 2016-06-20 by kitsunami