arduino/ArduinoCore-arc32 #181: CurieIMU get/set FreefallDetectionThreshold() implementations don't match datasheet

apiascik, 2016-05-11:
According to the [BMI160 datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMI160-DS000-07.pdf):

> low-g interrupt trigger threshold according to Val(int_low_th<7:0>) * 7.81 mg for Val(int_low_th<7:0>) >0 and 3.91 mg for Val(int_low_th<7:0>)=0. The range of the threshold is from 3.91 mg to 2.000 g; the default value corresponds to an acceleration of 375 mg 

Based on that, a value of 1 would make the threshold 7.81. Plugging in the default value (0x30), we get 374.88.

The implementation of getFreefallDetectionThreshold() and setFreefallDetectionThreshold(), however,  returns Val(int_low_th<7:0>) * 7.81 mg **+ 3.91** for Val(int_low_th<7:0>) > 0. This would mean that for  a value of 1, we would get 11.72 mg, and for the default value (0x30), we would get 378.79, which doesn't match the datasheet.

kitsunami, 2016-07-25:
Submitted this issue to the MIG documentation team, to review any datasheet updates needed.

piroozn, 2016-09-06:
The description for this problem shows that value matches the calculated value from the data sheet. To me the data sheet is showing the correct information and it is matching what the register value.
It is a bit confusing because they have to add the acceleration value to the initial acceleration value of 3.91 mg @ value=0. The data sheet is showing the value of 375 mg but is explaining that needs to be added to 3.91mg which makes it 374.88 (Rounded to 375) + 3.91 = 378.79 (Rounded to 379) as expected.

kitsunami, 2017-01-30:
Unable to duplicate issue.

closed 2017-01-30 by kitsunami