arduino/ArduinoCore-arc32 #183: setGyroOffset bug

dadez87, 2016-05-15:
I found a bug in "setGyroOffset" function.

`void CurieIMUClass::setGyroOffset(int axis, int offset)
{
    if (axis == X_AXIS) {
        setXGyroOffset(axis);
    } else if (axis == Y_AXIS) {
        setYGyroOffset(axis);
    } else if (axis == Z_AXIS) {
        setZGyroOffset(axis);
    }

    setGyroOffsetEnabled(true);
}`

Instead of "axis", "offset" should be written. Otherwise, the offset is set wrongly.

eriknyquist, 2016-05-15:
Hi @dadez87, if you look in https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieIMU/src/CurieIMU.cpp you'll see that these functions are using the 'offset' value. Perhaps you are looking at an old version?

eriknyquist, 2016-05-15:
@dadez87 looking at the history I can see this was fixed in February
https://github.com/01org/corelibs-arduino101/commit/e8b8a5db546696735d5de38e58253b75cb411e7a

If you take the latest version from Github then you won't have this issue.

kitsunami, 2016-05-27:
@eriknyquist, can you close this if it is fixed in current JSON releases?

closed 2016-06-20 by kitsunami