arduino/ArduinoCore-arc32 #184: Arduino 101 - McAffee reports trojan in arduino101load.exe

facchinm, 2016-05-16:
Issue opened by @VisualMicro in  https://github.com/arduino/Arduino/issues/4952

_Running windows and 1.6.9._
_I just installed the 1.0.5 Intel core. The previous 1.0.4 was ok._
_McAffee report attached._

[image]

bbaltz505, 2016-05-16:
I pulled the source for the 1.6.4+1.19-windows tag and built it. Initially, there were significant differences between the rebuilt binary and the one in GitHub but then I noticed many of them were a build path. Once I renamed the source folder to match the original build folder name then the rebuilt and released binaries match exactly.
This is built on an Ubuntu 12.04 LTS 64-bit with:
* go version go1.5.3 linux/amd64
* mingw-w64 2.0.1-1

facchinm, 2016-05-16:
It's obviously a false positive, sorry for not reporting it earlier :smile: 
The build from sources matches with the binary

VisualMicro, 2016-05-16:
Good news. Thanks for investigating so promptly. I will report it to McAffee.

calvinatintel, 2016-05-16:
The `.go` scripts in the [same directory](https://github.com/01org/intel-arduino-tools/tree/9282cb2ff5bb50f819bcd7f4af6adcb1f69e05b9/arduino101load) are the source for the `arduino101load.exe` for those who want to compile their own copy.

[VirusTotal scan](https://www.virustotal.com/en/file/4a0b2a35f6c7e1db393cc39d9be7c32c6b6c130c032c5ad470e1d15bef7316bf/analysis/1463415618/) detects 8 out of 56 matches which is rather low. Most hits say "proxy" and McAfee-GW-Edition says "BehavesLike.Win32.Downloader.wh". Since `arduino101load.exe` is used for communicating with the board, I suspect that a code snippet was incorrectly identified as a malicious uploader.

closed 2016-05-16 by calvinatintel