arduino/ArduinoCore-arc32 #185: Arduino101-specific #defines?

dantler, 2016-05-16:
I tried to port a library to Arduino101. This library was written for AVR and depended upon `<avr/interrupt.h>`, but I wanted to see if I could use `#ifdef` to swap in Arduino101-specific replacement routines.  I looked around but could not figure out which `#ifdef` to use!  I tried to export the cross-compiler's built-in #defines to see if I could detect the ARC architecture, but the export failed.

I ended up writing my own library which only works on Arduino101, but it still might be nice to use an `#ifdef` to warn users who aren't using Arduino101 that the library might not work.

Do you have a recommendation for an Arduino101-specific `#ifdef` that could be used for compatibility purposes?

intel-larry, 2016-05-16:
Try:

#if defined(__ARDUINO_ARC__)

dantler, 2016-05-17:
Thanks!  This works.  

In case anyone else looks at this, I found out that I could limit the scope of my Arduino library inside a [properties file](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification).  Arduino101 seems to use `arc32` as its architecture.

closed 2016-05-17 by dantler