arduino/ArduinoCore-arc32 #186: Scan Response capabilities missing

Damieneti, 2016-05-17:
It would be a great feature to add the BLE 4.0 Scan Response capabilities as described there: [http://www.argenox.com/bluetooth-low-energy-ble-v4-0-development/library/a-ble-advertising-primer/](url)

mjs513, 2016-06-01:
This may resolve a problem I am having with getting BLE to work with Processing3 using a BLED112 - not 100% that is the problem since it seems to work fine in Python itself.

eriknyquist, 2016-10-10:
@SidLeung has this issue been addressed by central add + peripheral re-factor?

noelpaz, 2016-10-11:
This was not implemented. Also the preview BLE from Arduino did not implement this. I asked @sgbihu and @SidLeung about this and I think the BLE stack does not currently support it as well.

Also note that Nordic does not really encourage this capability. We probably should close this and revisit it when we decide too and can do it (meaning we have to change the BLE stack).

sandeepmistry, 2016-10-11:
It would be great to get this in the next release. It would allow users to uses the full 31 bytes in the advertising data, then the scan response can be used for the local name.

> Also note that Nordic does not really encourage this capability. We probably should close this and revisit it when we decide too and can do it (meaning we have to change the BLE stack).

Do you have a link related to this?

kitsunami, 2017-02-24:
Relates to #412

closed 2017-03-08 by Damieneti