arduino/ArduinoCore-arc32 #190: millis() returns broken value if CurieIMU is accessed from interrupt context.

vshymanskyy, 2016-05-23:
https://forum.arduino.cc/index.php?topic=391836.0

Having same issue but  with extensive CurieBLE usage...

SidLeung, 2016-05-23:
Please provide Sketch output and the expected output.  Thanks.

vshymanskyy, 2016-05-26:
For example, millis() sometimes returns value which is less then previous calls, and then returns to normal on the next call.
This looks like interrupt locking problem, and is reproducible not only with ```millis()```.
This happens regularly when IMU or BLE is used frequently.

eriknyquist, 2016-06-20:
@vshymanskyy the millis function reads a counter that requires a regular interrupt to be updated.
In the sample code shown at the link above, CurieIMU.readMotionSensor is being called from interrupt context; This function, like most CurieIMU functions, involves communicating with the BMI160 via SPI, so it takes up a lot of time. So much time that you apparently miss a bunch of timer interrupts, while hanging around in interrupt context waiting for data from the IMU. I strongly recommend not to call CurieIMU functions from interrupt context.

eriknyquist, 2016-06-27:
@vshymanskyy can you show us some code please? without it, I will assume that you are doing the same thing that hsuenaga is doing on the forum link you provided (calling long-running / blocking IMU/BLE methods in interrupt context).

If this is the case, then a common solution is to modify your callback function so that all it does is set a global flag. Then in each iteration of `loop()`, you check if the flag has been set, and if so, you call your blocking BLE function, or whatever it is you need to do. In this way, you get out of interrupt context as quickly as possible and do the slow stuff during normal execution.

vshymanskyy, 2016-06-29:
@eriknyquist , yes I think I have a bit different case.
I'll try to make a minimal sketch showing the problem -> but it will take some time. I think a week or so. Thanks!

eriknyquist, 2016-06-30:
@vshymanskyy I'll be here!

eriknyquist, 2016-07-25:
@vshymanskyy any updates on this?

eriknyquist, 2016-08-08:
@vshymanskyy closing the issue due to a lack of response. Feel free to re-open if you ever return to it.

closed 2016-08-08 by eriknyquist