arduino/ArduinoCore-arc32 #191: BLECharacteristic.setValue(buff, len) blocks for 60-90 ms.

vshymanskyy, 2016-05-23:
Is such a big blocking time expected?

SidLeung, 2016-05-26:
The time taken for setValue to complete depends on the operation.  For instance, the BLE data transfer is done by notification.  A host is connected (subscribed) to your board and it is driving the connection interval at 60 msec.  When your application (Sketch) calls setValue(), the BLE stack can only transfer the data to the host on the next connection which is 60+ msec away.  If you average the time taken for a large amount of transfers, the average execution time would be between 60-90 msec.

bootchk, 2016-06-14:
Possibly the original question is: SHOULD the call block?  For example, in a different library, the Android BLE library, the documentation says their version of setValue() "sets the local value", and apparently it doesn't block.  And that library has another function, writeCharacteristic(), that also notifies any connected central client.  I could be wrong, I haven't used either library.

Another way of phrasing the question: "is setValue() synchronous IO?"  That is, does the call block until any connected, subscribing, central client receives a notification of the changed value?  Apparently, yes.

You would need to understand the architecture of the system and need to read much code to find the answer to that question.  Maybe this is a request for improved documentation, a line in the function's documentation that says "This call may block until any connected client is notified, which could take 90msec."

Or maybe this is a request for an enhancement, a function to set the value of the characteristic without blocking.  I'm not sure why you would want to do that, except maybe to keep a GUI responsive.  A calling app could use the behind the scenes RTOS to keep their app from becoming non-responsive when the library blocks.

SidLeung, 2016-06-14:
The setValue() routine is a blocking call.  The routine will not return until it has completed its task successfully or unsuccessfully.  setValue() first store the data for any Central to "read".  If there is subscription and "notification" is set, it will make use of the next BLE connection interval to send a frame of data to the Central.  Please note that notification is limited to 20 byte of data per call.  setValue() will return the control back to the caller upon transmission completion.
Please note that, as previous stated, the blocking time for a notification is not just determined by the execution of setValue().  The host (Central) determines the connection interval and that is an important factor in the setValue() blocking time.

eriknyquist, 2016-10-10:
Are we waiting for a response to move forward here? or can this be closed?

noelpaz, 2016-10-11:
We need to close this as Sidney has given the correct explanation. The connection interval can be tuned. Please bear in mind also the 20 byte limitation and you may have to go lower than 20 bytes if you have an aggressive connect interval

closed 2016-10-11 by vshymanskyy

sandeepmistry, 2016-10-11:
For clarification, is the Nordic stack blocking for 60 - 90 ms or is there something in the 101's nRF51822 firmware doing this?