arduino/ArduinoCore-arc32 #192: Error in member class CurieIMUClass::setDetectionDuration(int feature, float value) in CurieIMU.cpp

gianzalone77, 2016-05-24:
Hello,
I'm using the CURIE_IMU_ZERO_MOTION and I think there is an error at line 769 in CurieIMU.cpp file. Please see the case CURIE_IMU_ZERO_MOTION in the following code.
Regards,
Giuseppe

```
void CurieIMUClass::setDetectionDuration(int feature, float value)
{
    switch (feature) {
        case CURIE_IMU_FREEFALL:
            setFreefallDetectionDuration(value);
            break;

        case CURIE_IMU_SHOCK:
            setShockDetectionDuration(value);
            break;

        case CURIE_IMU_MOTION:
            setMotionDetectionDuration(value);
            break;

        case CURIE_IMU_TAP_SHOCK:
            setTapShockDuration(value);
            break;

        case CURIE_IMU_ZERO_MOTION:
            // ************************************************
            // setZeroMotionDetectionThreshold(value); 
            // The above line should be replaced with the following:  
            setZeroMotionDetectionDuration(value);
            break;

        case CURIE_IMU_TAP_QUIET:
            setTapQuietDuration(value);
            break;

        case CURIE_IMU_DOUBLE_TAP:
            setDoubleTapDetectionDuration(value);
            break;

        case CURIE_IMU_TAP:
        case CURIE_IMU_STEP:
        case CURIE_IMU_FIFO_FULL:
        case CURIE_IMU_DATA_READY:
        default:
            break;
    }
}
```

gianzalone77, 2016-05-26:
@jgithubs  I've found the same error also in member function:
CurieIMUClass::getDetectionDuration(int feature, float value)

SidLeung, 2016-06-01:
This issue is corrected in Bellatrix release.

gianzalone77, 2016-06-01:
It seems that the latest version still has the issue.

calvinatintel, 2016-06-02:
Hi @gianzalone77, by "latest version", do you mean the latest tip of the repo, or the latest release? A new release with the fix hasn't been published yet.

gianzalone77, 2016-06-02:
> A new release with the fix hasn't been published yet.

A.. ok! Thank you @calvinatintel, I meant the latest available on the repo.

closed 2016-07-11 by kitsunami