arduino/ArduinoCore-arc32 #201: DFU USB descriptor length doesn't match string

apiascik, 2016-05-31:
Not sure if this is the correct place to report this but...

In the file arduino101_firmware_source-v1/arduino101_firmware/bsp/bootable/bootloader/chip/intel/quark_se/defu_desc.c, lines 211-216:

```
{
		.bLength = sizeof("QRK_SE-Dev1.0") * 2,
		.bDescriptorType = UDESC_STRING,
		.bString = { 'A', 0, 'T', 0, 'P', 0, '-', 0, 'D', 0,
			'e', 0, 'v', 0, '1', 0, '.', 0, '0', 0 }
},
```

`.bLength` is based of the string `"QRK_SE-Dev1.0"`, whereas` bString` is set to `"ATP-Dev1.0"`.

calvinatintel, 2016-05-31:
@bigdinotech @SidLeung @eriknyquist Please review

eriknyquist, 2016-06-21:
Yep, that's a mistake alright, but we're not doing any firmware updates right?

kitsunami, 2016-07-07:
Not yet, but I'll set this on the next FW Update backlog.

kitsunami, 2016-12-19:
Changes to FW going forward will not have this.

closed 2016-12-19 by kitsunami