arduino/ArduinoCore-arc32 #205: The PWM do not work when gyro is working

vyuke, 2016-06-01:
hi, when i use an arduino101 to controlling LED by PWM via the internal IMU . i find PWM is not work,i could not figure it out ,could you help me?

SidLeung, 2016-06-01:
Would you please provide the following info:

1. Are you using the software PWM (CurireTime1PWM) or the hardware PWM.
2. How do the two modules, IMU and PWM, interact with each other.
3. How do the modules work together for the goal of your project.

Thanks.

eriknyquist, 2016-06-01:
@vyuke can you also provide the sketch, and give details about expected output vs. actual output, please.

eriknyquist, 2016-06-21:
@vyuke ping. Any updates?

eriknyquist, 2016-06-30:
@kitsunami recommend to close this in a week if still no response.

closed 2016-07-25 by kitsunami