arduino/ArduinoCore-arc32 #212: Linker using DCCM instead of SRAM

Mos-Fett, 2016-06-15:
In testing how much memory I could use on the Arduino 101 I found that I was limited to ~8KB instead of the 24KB of SRAM advertised. I opened a post on the Arduino forums explaining the details of my findings (linked here: http://forum.arduino.cc/index.php?topic=406411.0), but I was referred to the Intel Github after being told that it could be an issue with the linker using DCCM instead of SRAM. Do you know if the linker has this bug, or how I could fix it?

eriknyquist, 2016-06-15:
@Mos-Fett at first glance it looks like SRAM is indeed being used, in fact it doesn't look like DCCM is being used at all. You'll notice in that linker script that a start address for DCCM is define but it is never used. It may also be the case that SRAM is just full; a lot of it is used by core libraries to make your sketch work. I will investigate further and reply later today.

eriknyquist, 2016-06-15:
Short answer; no, this isn't a bug in the linker script, and there is indeed 24k of SRAM available. It's just full!

Longer answer; a lot of the SRAM is in use is in use to make your sketches work.
If we look at the linker script in variants/arduino_101/linker_scripts/flash.ld, you will notice that this 24k of SRAM contains; the stack, the heap, .bss section and .data sections.

I analysed the .map file for an empty sketch, just to see how much of the SRAM is used by corelibs. It was about 5k, with the .bss and .data sections combined. Next, if we look at the linker script, we can see there is 2k reserved for the stack (lines 50 and 170), and 8k reserved for the heap (lines 55 and 160).

so we start with 24k, but...

minus ~5k for corelibs
minus  ~2k for stack
minus ~8k for heap

And hey, what do you know, we have about ~8k left!

If you are really stuck for SRAM and you don't need 8k of heap space, you could try reducing the value of __HEAP_SIZE_MIN in the linker script, to gain back some SRAM.

Mos-Fett, 2016-06-16:
Interesting. Thanks @eriknyquist for the help, that is some really good information to have and certainly explains my question.

closed 2016-06-20 by kitsunami