arduino/ArduinoCore-arc32 #213: how to modify value of Characteristic

idone-wangqiang, 2016-06-16:
I want to modify the BLEWrite's value  from  0x02 to 0xFEC9

the wechat airsync protocol require the ble device  provide  Write & Indicate Characteristic,the Write Characteristic 's value must be 0xFEC9,and Indicate Characteristic's value must be 0xFEC7.

I have modify the BLECharacteristic.h file like this :
`enum BLEProperty {
    // broadcast (0x01) not supported
    BLERead						 = 0xFEC9,
    BLEWriteWithoutResponse		= 0x04,
    BLEWrite					= 0xFEC7,
    BLENotify				  = 0x10,
    BLEIndicate				  = 0xFEC8
};`

but it doesn't work

Can someone give me help?

327, 2016-06-20:
modify the uuid of Characteristic?

eriknyquist, 2016-06-20:
Hi @idone-wangqiang, you say "I want to modify the BLEWrite's value from 0x02 to 0xFEC9", but then in the code you show, you have set BLEWrite to 0xFEC7. As a result I'm not really sure what your question/problem is. Can you please do three things for me, so that I can help you;

1. Clearly state your intention, and show the code that you plan to use to achieve your intention.
2. State the expected outcome when you test your code changes (i.e. how will you know if it works)
3. State the actual outcome (How do you know that it didn't work).

idone-wangqiang, 2016-06-21:
Thank you very much for your reply.
My question is : How to modify the uuid of Characteristic in curieBLE library?

I don't know the default uuid  of Characteristic,but in arduino 101,

I want to set BLEWrtie's uuid to 0xFEC9,where can modify the uuid in curieBLE library ?

[airsync debugger](http://iot.weixin.qq.com/wiki/doc/blue/AirSyncDebugger2.3.0.apk)
This Android app can test the result.

eriknyquist, 2016-06-21:
@SidLeung do you know what the intention is here? Perhaps you can advise, in case I'm missing something.

@idone-wangqiang , If you would like to report a bug please follow the 3 steps I posted in my previous comment. If you have questions about how to use the BLE library then please direct them to the forum: https://forum.arduino.cc/index.php?board=103

closed 2016-07-22 by idone-wangqiang