arduino/ArduinoCore-arc32 #215: CurieBLE support for attribute length up to 512 bytes

LeAlesk, 2016-06-16:
The current length limitation may be blocking when trying to implement some Bluetooth services including utf8 formatted strings characteristics.

in BLECommon.h:
```
/* Theoretically we should be able to support attribute lengths up to 512 bytes
 * but this involves splitting it across multiple packets.  For simplicity,
 * we will just limit this to 20 bytes for now, which will fit in a single packet
 */
#define BLE_MAX_ATTR_DATA_LEN 20
```

SidLeung, 2016-06-16:
Please note that BLE Notification is limited to 1 BLE frame that can only hold 20 byte of user data and, thus, BLE_MAX_ATTR_DATA_LEN, is set to 20.  The #define can be changed to a higher limit (please keep it below 128) and send in all the data via the setValue() call.  There are two possible outcomes:
1.  The Central (host) read (poll for) the data out of the board (Peripheral), it can get all data.
2.  The Central uses Notification, the result is not clear.

mjs513, 2016-06-20:
There are some cases where you may want to send an attribute where the length is greater than 20.  I typically have string lengths that would to around 60. It would be nice to have a function call to change the attribute length without having to edit the API.  Avoids having custom mods to the corelibs.

Mike

SidLeung, 2016-06-20:
This feature is under development for the next release.  Stay tuned.

mjs513, 2016-06-24:
Thanks Sid.  Appreciate the info.

noelpaz, 2016-10-03:
Updates on this:  On a library level we have fix the ability to write up to 512 bytes. However found out that the BLE stack which is maintained by the thunder dome team , has issues when we a central is reading. There is a forced disconnect and the transaction never completes.

I filed a Thunderdome jira: FIRE-6872
And the ATL jira is 588

 as per @sgbihu 

Summary [ of what the Nordic FW currently supports ]
For peripheral
1. The stack only support read 100 bytes. Or the Nordic chip will give an NRF_ERROR_DATA_SIZE error.
2. The stack can support write 512 bytes.
For Central role.
1. The Nordic chip doesn't implement the BLE_GATT_OP_PREP_WRITE_REQ and BLE_GATT_OP_EXEC_WRITE_REQ.
2. Long read also not supported.for the Nordic Firmware 
3. Will not implement the central role example sketch.

closed 2017-02-21 by kitsunami