arduino/ArduinoCore-arc32 #216: CurieBLE: advertise while connected to another device

q89123003, 2016-06-16:
In BLE protocal, it is possible for a BLE peripheral to advertise data while connected to another central device.
Is it possible to modify the CurieBLE to allow BLE peripheral to at least keep advertise device name while connected to another central device?
As I trace  BLEPeripheral.cpp, I found that the private function _startAdvertising will return in the beginning if the _state != BLE_PERIPH_STATE_READY, and the _state will be changed into BLE_PERIPH_STATE_CONNECTED after the device is connected. So is it possible to modify this part to allow BLE peripheral advertise while connected?

Thanks!

SidLeung, 2016-06-16:
The reason for the Peripheral to stop advertising after a successful connection to a Central is to limit the connection to one Central at a time.  Not sure if, " allow[ing] BLE peripheral advertise while connected " is desirable.

q89123003, 2016-06-18:
Thanks for your reply.
I understand the reason. But I mean in BLE peripheral should be able to advertise unconnectable data while connected to central. However, CurieBLE seems not yet implemented the function of advertising data, therefore I say "at least" advertise device name.

SidLeung, 2016-06-20:
Please note that the Peripheral acts as if it has disappeared from the sense (going dark) once it is connected to a Central by stop advertising.  This will prevent other Centrals to discover it and attempt to connect.

kitsunami, 2016-10-03:
Closing per Sidney's comment above.

closed 2016-10-03 by kitsunami