arduino/ArduinoCore-arc32 #220: How do I set the option to use c99 on Arduino 101?

jimdgit, 2016-06-22:
I have extensive code that is in c99.
I have look thru the boards.txt file and can not find a way to turn on c99.
Note that no version of c++ can do what c99 does, so I really need to turn on c99 for this board.
Thanks, Jim

PaulStoffregen, 2016-06-23:
duplicate https://github.com/arduino/Arduino/issues/5063

jimdgit, 2016-06-23:
Paul, I was advised to post the issue here by facchinm, and he closed the issue so I would post it here.
So I don't really understand why you are marking it duplicate.
As for the example:
In  a file called test.c:
void f()
{
  for( int i = 0 ; i < 10 ;++i )
    ;
}
And the error from verify is:

'for' loop initial declarations are only allowed in C99 mode

Which is why I asked "how do I set c99 mode".
The problem is that I have this in 1000's of lines of code. So far, all other platforms have been able to compile in c99 mode.
Is it possible to set c99 mode for the 101?
Thanks for looking at this.

eriknyquist, 2016-06-23:
@jmidgit can you post the code that requires C99 to compile, and show the compile error? is it in a .cpp or .c file?

oqibidipo, 2016-06-23:
> I see you have a platform.txt for the 101, but I can't find it on my machine.

`Arduino15/packages/Intel/hardware/arc32/<version>/platform.txt`
`Arduino15` is where `preferences.txt` is, you can its location in the Preferences dialog.

eriknyquist, 2016-06-23:
OK, I backtracked over the issue on Arduino's github, saw your C file.
Open the file that @oqibidipo mentioned and add **-std=gnu99** to **compiler.c.flags**. See if that works for you.

jimdgit, 2016-06-23:
First thank you all for this information!
Arduino15/packages/Intel/hardware/arc32/<version>/platform.txt
On windows this is "hidden" in:
C:\Users\<your user name>\AppData\Local\
unlike the default ones that are in the install directory (that's why AI could not seem to locate these files)
I will try this tonight and post results.

jimdgit, 2016-06-23:
Ok, I can confirm that adding -std=gnu99 to compiler.c.flags. in Arduino15/packages/Intel/hardware/arc32//platform.txt solves all my issues.
Thanks again to @eriknyquist and @oqibidipo for taking time to help me out.

matthijskooijman, 2016-06-24:
Can we please fix this in the core itself? The compilation options should match those of the AVR and SAMD cores (which means using gnu11 and gnu++11 standards IIRC).

eriknyquist, 2016-06-24:
@matthijskooijman yes, there is `-std=gnu11` in the `compiler.c.flags` here:
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/platform.txt
So we should match that I guess. I'll look into it.

closed 2016-06-24 by eriknyquist