arduino/ArduinoCore-arc32 #226: How to expand SRAM?

tantra21, 2016-07-01:
I am using the 101 to drive Neopixel LED array. Each LED requires 3 bytes of RAM, and there are overall 20000 LEDs that I am working with. I am using the Adafruit Neopixel library to drive these Neopixel LEDs.  I am able to allocate memory for 1024 LEDs at a time and the Arduino 101 then runs out of memory. 
To solve this problem, I connected 1024 LEDs per GPIO pin on the Arduino 101 and I am allocating the memory for the LEDs connected to the individual GPIO pins during run time. This lets me work with 1024X14 (since there are 14 GPIOs) LEDs at a time, but every time I address LEDs connected to a particular GPIO pin I need to free the memory held by the LEDs connected to the a different GPIO pin, and reallocate this memory for the LEDs connected to the current GPIO pin. This causes the LEDs to flicker slightly. The program I am using is:
                                                                                                 
Adafruit_NeoPixel leds = Adafruit_NeoPixel();

Depending on the LED that I want to talk to, I set the length of the array and the pin as follows:
  
leds.updateLength(1024);
leds.updateType(NEO_GRB + NEO_KHZ800);
leds.setPin(pin_number);

This allocates memory for 1024 leds connected to the pin specified in the setPin function. The update length function frees memory(if any) held by the neopixels and allocates the memory for neopixels connected to the pin specified in the setPin function.

I wanted to know if it is possible to expand the SRAM or if there is any way I could drive 20000 LEDs using one Arduino 101 without the LEDs flickering?

eriknyquist, 2016-07-01:
hi @tantra21, it sounds like you need about 60k of RAM. Unfortunately there is only 24k out of 80k available to Arduino sketches (see details of the memory layout here https://github.com/01org/corelibs-arduino101/blob/master/variants/arduino_101/linker_scripts/flash.ld), and then even a large part of the 24k is in use to make your sketches work (see the discussion here https://github.com/01org/corelibs-arduino101/issues/212).

The remainder of the 80k is currently in use by the firmware & bootloader that run on the x86 core.

So, to summarise; right now your sketch has roughly 8k of SRAM to work with. You could try shrinking some memory areas in the linker script, as per my suggestions in the above linked issue, and this will gain you back a few k. But it will be impossible to get more than 24k, unless you modify the firmware.

intel-larry, 2016-07-01:
Hi,

There are limitations of course, with the Neopixel library. In particular, it uses the strategy of allocating all the LED memory on a per-led basis. The only way to use NeoPixel with larger strands than you have memory for is to spread them on multiple pins. This works, but you can run into problems of latency.

20,0000 LED’s on a single strand is quite a lot. There are various alternate strategies for dealing with large strands on Arduino boards that work around the memory limitations.

One approach you might want to look at is multiple 101 boards – and using something like SPI or UART to synchronize them for your display. This is how large scale lighting systems for stage and night clubs etc, work. A common protocol is Vixen.

Very large scale displays with this number of LED’s are usually driven by FPGA type devices.

Good luck with your project!

closed 2016-07-25 by kitsunami

PaulStoffregen, 2016-07-25:
This may be a little off-topic, but you might look at a library I wrote called OctoWS2811.  It's meant for controlling very large numbers of NeoPixel / WS2812 LEDs.  It needs different hardware (the off-topic part...) but the hardware is only a tiny fraction of the cost of so many LEDs.  Even then, 20000 LEDs is a huge number.  You'll probably need 4 or 5 boards working in parallel to control that many.  The FastLED library can use it as a driver, so if you need the sophisticated rendering features of FastLED, that might also be worth looking into.