arduino/ArduinoCore-arc32 #227: ARC_CTRL register inconsistent values

claudiaAl, 2016-07-01:
Hello!  I am experimenting with reading the contents of various registers and came across something weird.  I am not sure if it is just a simple programming issue that I overlooked... I hope so!  I am trying to read the contents of the ARC_CTRL register and get back two different values depending on if I am inside of a new function, or inside of my setup function.

```
uint32_t creg, saved;
void setup() {

  Serial.begin(115200);
  while(!Serial);

  adc_normal_wo_calib_REFERENCE();

  creg = READ_ARC_REG(ADC_CTRL);

  Serial.print("In SETUP ADC_CTRL after norm wo calib ");
  Serial.print(creg, BIN);
  Serial.println();

  disableADC();
  
  creg = READ_ARC_REG(ADC_CTRL);

  Serial.print("In SETUP ADC_CTRL after disableADC() ");
  Serial.print(creg, BIN);
  Serial.println();
}
```
And these are the values that I get.  The first two 32bit numbers come from the function that I call (disableADC()).  The function to read the register is the exact same in all cases.

```
ADC_CTRL after write: 10000000000000010101000000000011
ADC_CTRL in CREG after write: 10000000000000010101000000000011
In SETUP ADC_CTRL after norm wo calib 111100000100
ADC_CTRL reg before disabling ADC 111100000100
ADC_CTRL reg after disabling ADC 111100000100
In SETUP ADC_CTRL after disableADC() 111100000100
```

Does anyone know why the values may be different? And why it seems that in the read that happens in the setup() function, the value is truncated?

Any help or insight would be greatly appreciated.  Thanks!
--c

eriknyquist, 2016-07-01:
Hi @claudiaAl, this is difficult to answer since we can't see most of the reads/writes you're doing to ADC_CTRL. There are 5 lines of output, but I can only see one of those lines in your code. Can you please post a complete, verifiable example? i.e. an example that I can copy/paste into my own IDE and run myself.

claudiaAl, 2016-07-01:
Absolutely! My apologies :-)

```
#include <CurieTimerOne.h>
#include <SPI.h>

#define ATP_BLE_INT     5

uint32_t creg, saved;

void adc_normal_wo_calib_REFERENCE(){
      /* perform power up to "Normal mode w/o calibration" cycle if not already there */
    if( (creg & ADC_MODE_MASK) != ADC_NORMAL_WO_CALIB){

        /* Protect AR_IO_CREG_MST0_CTRL using lock and unlock of interruptions */
        saved = interrupt_lock();
        /* Read current CREG master */
        creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);
        creg &= ~(ADC_MODE_MASK);
        /* request ADC to go to Standby mode */
        creg |= ADC_STANDBY | ADC_CLOCK_GATE;
        WRITE_ARC_REG(creg, AR_IO_CREG_MST0_CTRL);
        interrupt_unlock(saved);
        /* Poll CREG Slave 0 for Power Mode status = requested status */
        while ( (creg = READ_ARC_REG(AR_IO_CREG_SLV0_OBSR) & 0x8) == 0);
        /* Protect AR_IO_CREG_MST0_CTRL using lock and unlock of interruptions */
        saved = interrupt_lock();
        creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);
        creg &= ~(ADC_MODE_MASK);
        /* request ADC to go to Normal mode w/o calibration */
        creg |= ADC_NORMAL_WO_CALIB | ADC_CLOCK_GATE;
        WRITE_ARC_REG(creg, AR_IO_CREG_MST0_CTRL);
        interrupt_unlock(saved);
        /* Poll CREG Slave 0 for Power Mode status = requested status */
        while ( ((creg = READ_ARC_REG(AR_IO_CREG_SLV0_OBSR)) & 0x8) == 0);
    }

    WRITE_ARC_REG(ADC_CLK_ENABLE | ADC_INT_DSB, ADC_CTRL);
    WRITE_ARC_REG(ADC_CONFIG_SETUP, ADC_SET);
    WRITE_ARC_REG(ADC_CLOCK_RATIO & ADC_CLK_RATIO_MASK, ADC_DIVSEQSTAT);

    
    Serial.print("ADC_CTRL after write: ");
    Serial.print(ADC_CTRL, BIN);
    Serial.println();

    creg = READ_ARC_REG(ADC_CTRL);
    Serial.print("ADC_CTRL in CREG after write: ");
    Serial.print(ADC_CTRL, BIN);
    Serial.println();
}

void disableADC(){

  creg = READ_ARC_REG(ADC_CTRL);

  Serial.print("ADC_CTRL reg before disabling ADC ");
  Serial.print(creg, BIN);
  Serial.println();

  saved = interrupt_lock();

  creg &= ~(1 << 2);

  creg |= READ_ARC_REG(ADC_CTRL);

  WRITE_ARC_REG(creg, ADC_CTRL);

  interrupt_unlock(saved);

  creg = READ_ARC_REG(ADC_CTRL);

  Serial.print("ADC_CTRL reg after disabling ADC ");
  Serial.print(creg, BIN);
  Serial.println();
}

void setup() {

  Serial.begin(115200);
  while(!Serial);

  adc_normal_wo_calib_REFERENCE();

  creg = READ_ARC_REG(ADC_CTRL);

  Serial.print("In SETUP ADC_CTRL after norm wo calib ");
  Serial.print(creg, BIN);
  Serial.println();

  disableADC();

   creg = READ_ARC_REG(ADC_CTRL);

  Serial.print("In SETUP ADC_CTRL after disableADC() ");
  Serial.print(creg, BIN);
  Serial.println();
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

I don't think my logic in disableADC() is correct yet, but I'm really looking at the difference in the register going from the adc_normal_wo_calib_REFERENCE() function back to setup()

Thanks in advance!

eriknyquist, 2016-07-01:
ADC_CTRL is a #define for the memory address of the ADC_CTRL register.

    #define ADC_CTRL                   (0x80015003)

So when you do this;

    Serial.print(ADC_CTRL, BIN);

You're not actually printing out the **value** stored at that address, you're printing the address itself. It's the same as doing this:

    Serial.print(0x80015003, BIN);

claudiaAl, 2016-07-01:
Arg dumb mistake! Thank you sir :-)

eriknyquist, 2016-07-01:
No problem.

closed 2016-07-01 by eriknyquist