arduino/ArduinoCore-arc32 #231: Bluetooth advertising does not start sometimes

bluetooth-mdw, 2016-07-07:
I'm testing implementing and testing various standard Bluetooth profiles on the Genuino 101. I'm proceeding in alphabetical order.... naturally. Alert Notification was OK. But... Automation IO revealed a problem which seemed to relate to descriptors. Basically the device does not start advertising. I stripped my code right down and isolated the problem to a degree.

@sandeepmistry went somewhat further and says "I've tracked it now to this line returning 0x285 instead or 0:

https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieBLE/src/BLECharacteristic.cpp#L205

"

Here's some code to illustrate the issue

```
// DOES NOT ADVERTISE

unsigned char AutomationIO_Digital_props = BLEWrite | BLEWriteWithoutResponse | 0;
unsigned char AutomationIO_Analog_props = BLEWrite | BLEWriteWithoutResponse | 0;
unsigned char AutomationIO_Aggregate_props = BLEWrite | BLEWriteWithoutResponse | 0;

// Services and Characteristics
BLEService AutomationIO("1815");
BLECharacteristic AutomationIO_Digital("2A56", AutomationIO_Digital_props, 20);
BLEDescriptor AutomationIO_Digital_CharacteristicPresentationFormat("2904", "0");

// Note I also tried:
// BLEDescriptor AutomationIO_Digital_CharacteristicPresentationFormat("2904", 0);
// and
// BLEDescriptor AutomationIO_Digital_CharacteristicPresentationFormat("2904", 0 , 1);
// and
// unsigned char buffer[7]={0x00,0x00,0x00,0x00,0x00,0x00,0x00};
// BLEDescriptor AutomationIO_Digital_CharacteristicPresentationFormat("2904", buffer, 7);

// GAP properties
char device_name[] = "My Device";

void setup() {

// set advertising packet content

  blePeripheral.setLocalName(device_name);

// add services and characteristics
  blePeripheral.addAttribute(AutomationIO);
  blePeripheral.addAttribute(AutomationIO_Digital);
  blePeripheral.addAttribute(AutomationIO_Digital_CharacteristicPresentationFormat);
// begin advertising
  blePeripheral.begin();
}
```

```
// ADVERTISES if I use a different descriptor instead
unsigned char AutomationIO_Digital_props = BLEWrite | BLEWriteWithoutResponse | 0;
unsigned char AutomationIO_Analog_props = BLEWrite | BLEWriteWithoutResponse | 0;
unsigned char AutomationIO_Aggregate_props = BLEWrite | BLEWriteWithoutResponse | 0;

// Services and Characteristics
BLEService AutomationIO("1815");
BLECharacteristic AutomationIO_Digital("2A56", AutomationIO_Digital_props, 20);
BLEDescriptor tempDescriptor = BLEDescriptor("2901", "Temp Celsius");

// GAP properties
char device_name[] = "My Device";

void setup() {

// set advertising packet content

 blePeripheral.setLocalName(device_name);

// add services and characteristics
  blePeripheral.addAttribute(AutomationIO);
  blePeripheral.addAttribute(AutomationIO_Digital);
  blePeripheral.addAttribute(tempDescriptor);
// begin advertising
  blePeripheral.begin();
}
```

eriknyquist, 2016-10-10:
@bluetooth-mdw, since you posted this issie, the BLE central feature has been added, and the peripheral code has also been re-factored. Are you still seeing this issue?

bluetooth-mdw, 2016-10-11:
@eriknyquist Hi, I suspect I'm running exactly the same library version I was when I first bought the Arduino 101. What library version is the latest? I'll see if I can update and re-test. Thanks for the info.

eriknyquist, 2016-10-13:
@bluetooth-mdw just get the latest version from the IDE boards manager.

kitsunami, 2017-02-21:
The information for updating to the latest BLE FW is here: http://forum.arduino.cc/index.php?topic=443728.0

Can you please retest @bluetooth-mdw  ?

closed 2017-03-13 by kitsunami