arduino/ArduinoCore-arc32 #232: Register Address confusion

claudiaAl, 2016-07-07:
Hello!  I am going through trying to disable the SPI bus and am diving into the SPI control and enable registers.  I am using the SPIClass.end() function as my reference for what I need to do to accomplish this.  I am trying to understand what bits are being changed and have confused myself... can anyone clarify this for me?  Thanks in advance!

```
scss_registers.h
#define PERIPH_CLK_GATE_CTRL            (SCSS_REGISTER_BASE + 0x018)
/* SPI */
#define SOC_MST_SPI1_REGISTER_BASE      (0xB0001400)
```

SPIClass.h only touches SPI1 so I am assuming that SPI0 is what the BMI160 is using.
So if we look at the bit that the clock gate control is being defined as it is 0x018 = 24.  However, the documentation shows that it should be at bit 28 (or 29 if we are starting at 1).

> 2.1.6.86 CREG register detailed description
> 2.1.6.87 IO_CREG_MST0_CTRL (IO_CREG_MST0_CTRL)
>      MEM Offset(80018000)
> 
> 31     AD_CLK_GATE
> 30     I2C1_CLK_GATE
> 29     I2C0_CLK_GATE
> 28     SPI1_CLK_GATE
> 27     SPI0_CLK_GATE
> 26:20   ADC CALIBRATION VALUE

The specific code that I am looking at lives in SPI.cpp (SPIClass.end())

```
SPI1_M_REG_VAL(SPIEN) &= SPI_DISABLE;
MMIO_REG_VAL(PERIPH_CLK_GATE_CTRL) &= DISABLE_SPI_MASTER_1;
```
Disabling the SPI seems ok, but when we disable the clock, I'm confused as to what bit is being changed.  Am I missing something obvious here?

eriknyquist, 2016-07-07:
Hi @claudiaAl, 

`#define PERIPH_CLK_GATE_CTRL            (SCSS_REGISTER_BASE + 0x018)`

This is actually a memory location, rather than a bitmask (it's the address of the PERIPH_CLK_GATE_CTRL register).

so, when we look at this line;

`MMIO_REG_VAL(PERIPH_CLK_GATE_CTRL) &= DISABLE_SPI_MASTER_1;`

it's saying "access the memory location PERIPH_CLK_GATE_CTRL, and apply the bitmask DISABLE_SPI_MASTER_1 to the value that lives there"

As to what bit is being changed, look in SPI_registers.h

```
#define     ENABLE_SPI_MASTER_0   (0x1 << 14)
#define     DISABLE_SPI_MASTER_0  (~ENABLE_SPI_MASTER_0)
#define     ENABLE_SPI_MASTER_1   (0x1 << 15)
#define     DISABLE_SPI_MASTER_1  (~ENABLE_SPI_MASTER_1)
```

So apparently it's bit 14 for SPI0, and bit 15 for SPI1

eriknyquist, 2016-07-07:
But-- an unrelated question-- is there any particular reason you need to re-implement `SPI.end()`? why can't you just call the existing method to disable the SPI bus?

claudiaAl, 2016-07-07:
Good question.  I can.  That's actually where I started, but I didn't see any change in my current readings, so in an effort to be as thorough as possible, I wanted to go in and read the registers and understand what was really happening.  If this all eventually goes into a library, I am assuming I would just use my end() function rather than duplicating the code...?

eriknyquist, 2016-07-07:
like I said, I think SPI will be disabled by default anyway until you call `*.begin()`. So calling `SPI.end()` without first calling `SPI.begin()` is effectively changing nothing.

claudiaAl, 2016-07-07:
got it!  thank you!

claudiaAl, 2016-07-07:
ok, so I have played with calling begin and end in order to see what's happening in the register.  It's just for my own learning and to prove to myself that I can see the changes.  I'm not sure that I am reading the registers correctly though.  I think I am specifically interested in bit 0 (SPIEN) of the SPI enable register because from what I understand, that will enable/disable all SPI master operations.  I was hoping to see that bit flip between calls, but I only see it go from 0 to 1 and never back to 0... which makes me question if I am reading the register correctly...

in my sketch I call:
`creg = SPI1_M_REG_VAL(SPIEN);`

I referenced SPI_registers.h which defines:
```
/* Macro to access SPI1 Master controller register offset */
#define SPI1_M_REG_VAL(reg) \
    MMIO_REG_VAL_FROM_BASE(SOC_MST_SPI1_REGISTER_BASE, (reg))
```

Am I reading this incorrectly?  Thank you!!

eriknyquist, 2016-07-08:
Do you have the latest version? I do, and I can't see `SPI1_M_REG_VAL`, only `SPI_M_REG_VAL`.
Maybe you have an old version?

You have the right idea, though; if you are using the latest, then you can read it like this;

    creg = SPI_M_REG_VAL(SOC_MST_SPI1_REGISTER_BASE, SPIEN);

But, there is a sneaky twist here; SPI (i.e. `SPI.begin()`) actually uses the **hardware** SPI1 (i.e. SOC_MST_SPI1_REGISTER_BASE). So when I added support for SPI1, the remaining SPI controller is unfortunately named SPI0. The result is that `SPI.begin()` accesses `SOC_MST_SPI1_REGISTER_BASE`, and `SPI1.begin()` accesses `SOC_MST_SPI0_REGISTER_BASE`. You can see this in SPI.cpp:

    SPIClass SPI(SPIDEV_1);
    SPIClass SPI1(SPIDEV_0);

and SPI.h:

    #define SPIDEV_0 0
    #define SPIDEV_1 1

    ...

    int spidevs[NUM_SPIDEVS][4] =
    {
        /* base addr.                     Clk. enable value    Clk. disable value    SS GPIO */
        {(int)SOC_MST_SPI0_REGISTER_BASE, ENABLE_SPI_MASTER_0, DISABLE_SPI_MASTER_0, SPI0_CS},
        {(int)SOC_MST_SPI1_REGISTER_BASE, ENABLE_SPI_MASTER_1, DISABLE_SPI_MASTER_1, SPI1_CS}
    };

So, you would test the **Arduino** SPI1 instance like this;

    /* enable SPI1 */
    SPI1.begin()

    /* Read SPIEN */
    creg = SPI_M_REG_VAL(SOC_MST_SPI1_REGISTER_BASE, SPIEN);

    /* Disable SPI1 */
    SPI1.end()

    /* Read SPIEN */
    creg = SPI_M_REG_VAL(SOC_MST_SPI1_REGISTER_BASE, SPIEN);

eriknyquist, 2016-07-08:
I just edited my last post- I had mistakenly used `SOC_MST_SPI1_REGISTER_BASE` in the final example. It's correct now :)

eriknyquist, 2016-07-08:
And, I was answering the original question I saw in the email notification-- I didn't notice you had removed the question. Did you figure it out?

claudiaAl, 2016-07-08:
I think I am close... I ended up using this after referencing SPI_registers.h
     `creg = SPI1_M_REG_VAL(CTRL0);`

I just compared the two versions and it looks like the definition changed from 

```
/* Macro to access SPI1 Master controller register offset */
#define SPI1_M_REG_VAL(reg) \
    MMIO_REG_VAL_FROM_BASE(SOC_MST_SPI1_REGISTER_BASE, (reg))
```
to
```
/* Macro to access SPI Master controller register offset */
#define SPI_M_REG_VAL(base, reg) MMIO_REG_VAL_FROM_BASE(base, (reg))
```
But I think that they do the same thing...

eriknyquist, 2016-07-08:
CurieIMU class uses SOC_MST_SPI1_REGISTER_BASE (i.e. `SPI.begin()`), but this SPI controller is only initialised by `CurieIMUClass::begin()`

Honestly, I don't  think there's any point in re-implemnting the `SPI.end()` method for your low power lib. Simply not calling `SPI.begin()` is already enough to ensure that the SPI controller is disabled.

claudiaAl, 2016-07-08:
Can we look at a quick use case?  Let's say we have a sketch running that initializes the CurieIMU class because we want to be collecting information from the sensors... let's say we are focusing on the motion and temperature sensors to determine if a device is being worn.  If we determine that the device is not being worn, I would like to disable the SPI bus (assuming that we can verify that disabling it will indeed provide power savings).  It would then need to be woken up once we see that the device is being worn again... no clue how I'm doing this yet... baby steps!

This means that my SPI1 would be needing to go between begin() and end() calls, but my SPI0 can remain disabled the whole time... is that correct? Am I making this harder than it needs to be?

eriknyquist, 2016-07-08:
For sanity's sake, let's just talk about SPI and SPI1 (i.e. Arduino's `SPI.begin()` and `SPI1.begin()`) from now on, even though we know now it's really SPI1 and SPI0 behind the scenes.

So for your use case, it sounds like you basically need some kind of `CurieIMUClass::end()` method, which doesn't exist right now. Maybe instead of messing with SPI, we should be thinking about adding an `end()` method. @SidLeung @bigdinotech what do you think?

claudiaAl, 2016-07-08:
It doesn't look like `CurieIMU.begin()` writes anything to the SPI Master 0 register.  When I call `SPI.begin()` and `SPI.end()` I can see changes made in the SPI1 Register (SPIEN, CTRL0, etc.).  When I call `SPI1.begin()` and `SPI1.end()` I see changes in the SPI0 Register as expected.  The `CurieIMU.begin()` method calls the `ss_spi_init()` function which sets the peripheral to inactive so the clock can be gated, disables the SPI1 controller, and enables the controller clock to allow for register writes.  That's followed by the BMI160 initialization.  But nothing ever changes in the control register, and there are a number of things that the `SPIClass.begin()` function does that the CurieIMU initialization doesn't.  

I've confused myself... should I just be working with the SPI class??

eriknyquist, 2016-07-11:
@claudiaAl I was wrong, there are actually 3 SPI controllers in use. 1 goes to pins 10-13 (`SPI.begin()`), one goes to the onboard SPI flash (`SPI1.begin()`), and the third is used only for the BMI160 and is initialised with `ss_spi_init()`. So, you want to look at the implementation of `ss_spi_init()`, I guess.

I'm also going to implement a `CurieIMU.end()` method, to disable the 3rd SPI controller.

eriknyquist, 2016-07-11:
Try this change @claudiaAl  https://github.com/01org/corelibs-arduino101/pull/235

claudiaAl, 2016-07-21:
Thank you Erik!  I was able to successfully test the running and disabling of all three SPI buses.  Interestingly enough, SPI and SPI1 behave as expected... running begin() and not doing anything consumes more energy than having it disabled.  However, the CurieIMU.end() method causes more power consumption than its begin() method due to the setting of some pins.

closed 2016-07-21 by claudiaAl

eriknyquist, 2016-07-22:
"However, the CurieIMU.end() method causes more power consumption than its begin() method due to the setting of some pins"

Yes, that's why an Arduino101 board is really not the best option for testing your power savings library, since there are a lot of power-hungry components on the board that you can't turn off. Even if you take the whole Curie chip off, I'm sure there will still be a significant current draw from the rest of the board.