arduino/ArduinoCore-arc32 #234: Change in platform.txt won't allow compilation

claudiaAl, 2016-07-08:
Hello!  I am working on another issue, and decided to update my version.  I was on 1.0.3 and I just realized y'all are on 1.0.6.  I followed the directions on the git page and tried to compile my sketch, as well as the example Blink sketch.  In both cases I get this error:

> arc-elf32-g++: error: unrecognized argument in option ‘-mcpu=quarkse_em’
> arc-elf32-g++: note: valid arguments to ‘-mcpu=’ are: ARC600 ARC601 ARC700 ARCv2EM ARCv2HS arc600 arc601 arc700 arcem archs
> exit status 1
> Error compiling for board Arduino/Genuino 101.

It looks like the issue may be in the platform.txt file, but I really don't know... is this a known problem? 

Thanks!

bbaltz505, 2016-07-08:
Hi.
The new platform.txt has been updated to use a new ARC toolchain. Replace your current one with one of these: 
* https://downloadmirror.intel.com/25470/eng/arc-toolchain-windows-arcem-1.0.1.zip
* https://downloadmirror.intel.com/25470/eng/arc-toolchain-osx-arcem-1.0.1.tar.bz2
* https://downloadmirror.intel.com/25470/eng/arc-toolchain-linux32-arcem-1.0.1.tar.bz2
* https://downloadmirror.intel.com/25470/eng/arc-toolchain-linux64-arcem-1.0.1.tar.bz2

claudiaAl, 2016-07-08:
Thank you @bbaltz505 !  I took the contents of the arc32 folder from the link that you provided, and replaced the contents of my existing **Arduino15 > packages > Intel > tools > arc-elf32  > 1.6.4+1.0** folder.  I'm not sure if the version number mismatch might cause a problem in the future, but I am able to successfully compile my Blink example.  Thanks again!

eriknyquist, 2016-07-08:
@claudiaAl, you can get 1.0.6 from the IDE boards manager. Everything should just work if you do it that way.

claudiaAl, 2016-07-08:
I went ahead and reverted to 1.6.10, and then updated back to 1.6.11.  It shows it being installed, however the version number of the directory (1.6.4+1.0) did not change.  Everything seems to be working just fine though. Thanks you guys!

bbaltz505, 2016-07-11:
@claudiaAl, glad to hear things are working. It might be cleaner to start with a clean IDE folder. If you want, you can delete the Curie package folder in the IDE:
%localappdata%\Arduino15\packages\Intel    (in Windows)

This will wipe out all installed Galileo, Edison and Arduino101 packages bu the you can install a clean build.

If everything is fine, you can close the issue.
Thanks!

claudiaAl, 2016-07-14:
Hello again!  In attempt to start from a clean slate, I removed everything... and then installed the latest version of the IDE (1.6.9 for MacOS).  I tried to install the Curie package, but as soon as I opened up the Board Manager, I received an error stating 

> Error downloading http://downloads.arduino.cc/packages/package_index.json

If I ignore that message and attempt to download the Curie package anyway, I am only allowed to install versions 1.0.4 and 1.0.5, and I get this error:

> Error downloading http://github.com/01org/corelibs-arduino101/archive/1.6.4+1.68.zip

(I noticed that the name of that file is not the same as what gets downloaded as a zip directly from the repo)

I then removed everything again, and tried the nightly build.  Same thing.  Although with this build, I am given the option of the 1.0.6 Curie package version.

I bet I can figure out how to manually install everything, but has anyone else run into this problem?  Thanks in advance!

bbaltz505, 2016-07-14:
@claudiaAl, that JSON link is the default package list from Arduino so I'm not sure why it failed to load. Is it a valid URL using your web browser? Since that list did not update, the newer version will not be available in Board Manager. Are you behind a firewall or proxy perhaps? 

Also, when you say you "removed everything", did that include the "/Users/<username>/Library/Arduino15" folder?

intel-larry, 2016-07-14:
Hi Claudia - is there any chance you're working on some kind of corporate network?  We've found that the failure of the Boards Manager to download the file is often due to problems with proxy servers on corporate, university and some other managed networks.  For instance, inside of Intel, due to the high security on our network, we have to manually enter the proxy server in the IDE preferences panel (network tab).

claudiaAl, 2016-07-14:
@bbaltz505 and @intel-larry, it was the Intel network.  I manually entered the proxy server and all is well.  Thank you!!

closed 2016-07-14 by claudiaAl

bbaltz505, 2016-07-14:
Ah, good. If you wiped out your preferences, any previous proxy setting would have been lost.