arduino/ArduinoCore-arc32 #246: I2C Control Question

claudiaAl, 2016-07-21:
Morning!

I am looking into what it takes to disable the I2C bus and am referencing the function `DRIVER_API_RC ss_i2c_clock_disable(I2C_CONTROLLER controller_id)` in `ss_i2c_iface.c`.  The function looks like this, and I am interested in what the two `WRITE_ARC_REG` functions are doing.

```
DRIVER_API_RC ss_i2c_clock_disable(I2C_CONTROLLER controller_id)
{
    i2c_info_pt dev = &i2c_master_devs[SS_CTRL_ID(controller_id)];
    uint32_t saved;

    /* Protect registers using lock and unlockl of interruptions */
    saved = interrupt_lock();

    /* disable clock to peripheral */
    WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(AR_IO_CREG_MST0_CTRL)) & ~( 1 << dev->creg_i2c_clk_ctrl),
                                   (volatile uint32_t)(AR_IO_CREG_MST0_CTRL));
    /* disable device */
    WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(dev->reg_base+I2C_CON)) & ~(I2C_ENABLE_MASTER),
                                   (volatile uint32_t)(dev->reg_base+I2C_CON));
    interrupt_unlock(saved);
    return DRV_RC_OK;
}
```
The first looks like it is writing to the CREG control register, in order to disable the clock.  It takes a 1, shifts it to the left until it gets to either the I2C0, or I2C1 control bit, negates it, and then it is ANDed with what was originally in the CREG register.  

The second function changes the I2C Control Register.  It uses `I2C_ENABLE_MASTER` which is defined in `i2c_priv.h` as:
`#define     I2C_ENABLE_MASTER       (1 << 0)`
So we are putting a 1 in the 0 bit, turning it into a 0 (which disables I2C), ANDing it with the control register, and writing it to the control register.  

I got confused when I noticed that my bits were not being flipped as I had intended in the CREG control register.  According to documentation for the register:

> (Bit 30) I2C1_CLK_GATE (I2C1_CLK_GATE)
> When I2C1_CLK_GATE is 1 it indicates that the I2C 1 peripheral is inactive and its clock can be gated.

It seems to me that the first write will always write a '0' to the bit(s) that disables one of the I2C clocks, however, I thought that it needed to be a '1' to disable... 
The second write does the same thing, but I am assuming here that because we are looking at an 'ENABLE' bit, that 0 is disabled, and 1 is enabled... we are good.

Any thoughts on this?  Did I just come up with a fuzzy explanation to confuse myself?  I'm trying to understand what exactly I need to write to the registers to disable the bus but this is the opposite of what I had worked out... thanks in advance!

Sorry for the long explanation, helps me work it out :-)

bigdinotech, 2016-07-21:
Hi @claudiaAl 

Can you share some code I can try it out on my side?

Also,
Are you modifying those files directly inside the drivers folder inside your corelibs then compiling using the IDE?

If so you may have an extra step. You will need to generate a libarc32drv_arduino101.a which resides inside variants/arduino101.

To do so you will have to run the make file inside system/libarc32_arduino101
using the arc toolchain

I did something like this:
export PATH=~/.arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1/bin/:$PATH
make

claudiaAl, 2016-07-21:
Hi @bigdinotech , thanks for the response.  I am actually writing everything from the Arduino IDE and testing that way.  Here is a bit of code, please excuse its ugliness.. 

```
#include <CurieTimerOne.h>
#include <CurieIMU.h>
#include <SPI.h>  
#include <CurieI2S.h>
#include <portable.h>
#include <eiaextensions.h>
#include <i2c_priv.h>

uint32_t creg, saved;
int time;
const int oneSecInUsec = 1000000;  

void disableCurieTimerOne(){
  CurieTimerOne.kill();
}
void timedBlinkIsr()   // callback function when interrupt is asserted
{
//  digitalWrite(13, toggle);
//  toggle = !toggle;  // use NOT operator to invert toggle value
}

void gateAllClocks(){
  
  creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);

  Serial.print("Master creg before clock gating: ");
  Serial.print(creg, BIN);
  Serial.println();

  saved = interrupt_lock();

  creg = (READ_ARC_REG(AR_IO_CREG_MST0_CTRL) | (SPI0_CLOCK_GATE | SPI1_CLOCK_GATE | I2C0_CLOCK_GATE | I2C1_CLOCK_GATE | ADC_CLOCK_GATE));

  Serial.print("creg before the clock gate write: ");
  Serial.print(creg, BIN);
  Serial.println();
  
  WRITE_ARC_REG(creg, AR_IO_CREG_MST0_CTRL);

  interrupt_unlock(saved);

  creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);

  Serial.print("Master creg after gating clocks: ");
  Serial.print(creg, BIN);
  Serial.println();  
}
void disableI2C0(){
   // if(!(AR_IO_CREG_MST0_CTRL+CREG_CLK_CTRL_I2C0)){
    //DISABLE THE CLOCK AND I2C0
      /* Protect registers using lock and unlockl of interruptions */
    saved = interrupt_lock();

    creg = READ_ARC_REG((volatile uint32_t)AR_IO_CREG_MST0_CTRL); 
    creg |= ( 1 << CREG_CLK_CTRL_I2C0);

    WRITE_ARC_REG(creg, (volatile uint32_t)(AR_IO_CREG_MST0_CTRL));

    /* disable clock to peripheral */
//    WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(AR_IO_CREG_MST0_CTRL)) & ~( 1 << CREG_CLK_CTRL_I2C0),
//                                   (volatile uint32_t)(AR_IO_CREG_MST0_CTRL));

    /* disable device */
    WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(AR_IO_I2C_MST0_CON+I2C_CON)) & ~(I2C_ENABLE_MASTER),
                                   (volatile uint32_t)(AR_IO_I2C_MST0_CON+I2C_CON));

    interrupt_unlock(saved);
   // }
}

void disableI2C1(){

    //DISABLE THE CLOCK AND I2C0
      /* Protect registers using lock and unlockl of interruptions */
    saved = interrupt_lock();

    /* disable clock to peripheral */
    WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(AR_IO_CREG_MST0_CTRL)) & ~( 1 << CREG_CLK_CTRL_I2C1),
                                   (volatile uint32_t)(AR_IO_CREG_MST0_CTRL));

    /* disable device */
    WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(AR_IO_I2C_MST1_CON+I2C_CON)) & ~(I2C_ENABLE_MASTER),
                                   (volatile uint32_t)(AR_IO_I2C_MST1_CON+I2C_CON));

    interrupt_unlock(saved);
}

void setup() {

  Serial.begin(115200);
  while(!Serial);

   gateAllClocks();

  creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);

  Serial.print("Master creg init: ");
  Serial.print(creg, BIN);
  Serial.println();  

  creg = READ_ARC_REG(AR_IO_I2C_MST0_CON+I2C_CON);

  Serial.print("AR_IO_I2C_MST0_CON+I2C_CON init: ");
  Serial.print(creg, BIN);
  Serial.println(); 

  disableI2C0();

  creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);

  Serial.print("Master creg after disabling I2C0: ");
  Serial.print(creg, BIN);
  Serial.println();  

  creg = READ_ARC_REG(AR_IO_I2C_MST0_CON+I2C_CON);

  Serial.print("I2C0 Master after disabling: ");
  Serial.print(creg, BIN);
  Serial.println();  
  
  //disableI2C1();
  
  // put your setup code here, to run once:
  //CurieTimerOne.initialize(50000);
  //disable TimerOne
  //CurieTimerOne.kill();

  //adc_deep_power_down();
  //adc_power_down();
  //adc_standby();
  //adc_normal_calib();
  //adc_normal_wo_calib();

  //CurieI2S.end();
}

void loop() {
  // put your main code here, to run repeatedly:

    //time = oneSecInUsec / 2; // time is used to toggle the LED is divided by i
    //CurieTimerOne.start(time, &timedBlinkIsr); 
  //CurieTimerOne.kill();
  //CurieTimerOne.start();

}
```
The `WRITE_ARC_REG` function that is commented out is what I took from the driver thinking that I could reuse the code from the driver disabling a device.  I call `gateAllClocks()` first so that I can make sure to see the last 5 bits set in the CREG register.  Then I call `disableI2C0()` thinking that the '1' in the I2C0 clock bit would remain... however when I use the driver code it turns it to a 0, meaning (what I think is) enabled.  I entered 3 lines of code above that commented out function with what I think is the correct logic to see if I would change the bit.  But I'm not really sure if I'm right or not.  I haven't worked out reading the IC2 control register yet so I'm not positive that's correct yet...

claudiaAl, 2016-07-21:
I'm going through my code and I'm not convinced of what I am doing... I'm still wondering about the driver code, but I'm reviewing what I have posted and am doing a few more tests to make sure my logic is right.  I'll repost later today, thanks!

claudiaAl, 2016-07-21:
Ok, so I have gone through and double checked my Arduino sketch to make sure that I was actually reading the values in my registers.  Now that I know how to access what I need, I am still confused as to whether I need to be setting bits 29 and 30 of the CREG master register to a '1' or to a '0' when disabling the I2C0 and I2C1 clocks respectively. If it is indeed a '1' as the documentation states, then I don't understand why in the driver code, `ss_i2c_clock_disable` is setting the bits to '0' rather than '1'.  

Hopefully my question makes better sense now... thanks!

eriknyquist, 2016-07-25:
HI @claudiaAl , can you please post some code we can try which demonstrates your question?

claudiaAl, 2016-07-25:
Hey Erik!  So here is a sketch that initially does exactly what the `initADC()` method does in `variant.cpp` so that I can see the entire register with a few bits set.  I then want to go in and disable I2C0.  But I am not sure if setting it to a '1' is actually disabling it or if I should be setting it to '0'.  I think I am misunderstanding the documentation... 

```
#include <portable.h>
#include <i2c_priv.h>

uint32_t creg, saved;

void adc_normal_wo_calib(){

    uint32_t creg;
    uint32_t saved;

    /* read creg slave to get current Power Mode */
    creg = READ_ARC_REG(AR_IO_CREG_SLV0_OBSR);

    /* perform power up to "Normal mode w/o calibration" cycle if not already there */
    if( (creg & ADC_MODE_MASK) != ADC_NORMAL_WO_CALIB){

        /* Protect AR_IO_CREG_MST0_CTRL using lock and unlock of interruptions */
        saved = interrupt_lock();
        /* Read current CREG master */
        creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);
        creg &= ~(ADC_MODE_MASK);
        /* request ADC to go to Standby mode */
        creg |= ADC_STANDBY | ADC_CLOCK_GATE;
        WRITE_ARC_REG(creg, AR_IO_CREG_MST0_CTRL);
        interrupt_unlock(saved);
        /* Poll CREG Slave 0 for Power Mode status = requested status */
        while ( (creg = READ_ARC_REG(AR_IO_CREG_SLV0_OBSR) & 0x8) == 0);
        /* Protect AR_IO_CREG_MST0_CTRL using lock and unlock of interruptions */
        saved = interrupt_lock();
        creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);
        creg &= ~(ADC_MODE_MASK);
        /* request ADC to go to Normal mode w/o calibration */
        creg |= ADC_NORMAL_WO_CALIB | ADC_CLOCK_GATE;
        WRITE_ARC_REG(creg, AR_IO_CREG_MST0_CTRL);
        interrupt_unlock(saved);
        /* Poll CREG Slave 0 for Power Mode status = requested status */
        while ( ((creg = READ_ARC_REG(AR_IO_CREG_SLV0_OBSR)) & 0x8) == 0);
    }

    WRITE_ARC_REG(ADC_CLK_ENABLE | ADC_INT_DSB, ADC_CTRL);
    WRITE_ARC_REG(ADC_CONFIG_SETUP, ADC_SET);
    WRITE_ARC_REG(ADC_CLOCK_RATIO & ADC_CLK_RATIO_MASK, ADC_DIVSEQSTAT);
}

void disableI2C0(){
  
    creg = READ_ARC_REG((volatile uint32_t)AR_IO_CREG_MST0_CTRL);

    if(!(creg & (1 << CREG_CLK_CTRL_I2C0))){
      //DISABLE THE CLOCK AND I2C0
      /* Protect registers using lock and unlockl of interruptions */
      saved = interrupt_lock();

      //read the CREG Master register
      creg = READ_ARC_REG((volatile uint32_t)AR_IO_CREG_MST0_CTRL); 

      Serial.print("Inside disableI2C0() AR_IO_CREG_MST0_CTRL: ");
      Serial.print(creg, BIN);
      Serial.println();  

      //Read the I2C MST0 Register
      creg = READ_ARC_REG((volatile uint32_t)AR_IO_I2C_MST0_CON); 

      Serial.print("Inside disableI2C0() AR_IO_I2C_MST0_CON: ");
      Serial.print(creg, BIN);
      Serial.println();  

      //read the CREG Master register   
      creg = READ_ARC_REG((volatile uint32_t)AR_IO_CREG_MST0_CTRL); 
      //set the I2C0 clock gate bit and OR it in
      creg |= ( 1 << CREG_CLK_CTRL_I2C0);
    
      Serial.print("creg after I2C control bit set: ");
      Serial.print(creg, BIN);
      Serial.println();  

      WRITE_ARC_REG(creg, (volatile uint32_t)(AR_IO_CREG_MST0_CTRL));

      creg = READ_ARC_REG((volatile uint32_t)AR_IO_CREG_MST0_CTRL); 

      Serial.print("After write to AR_IO_CREG_MST0_CTRL: ");
      Serial.print(creg, BIN);
      Serial.println();    

      /* disable clock to peripheral */
   // WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(AR_IO_CREG_MST0_CTRL)) & ~( 1 << CREG_CLK_CTRL_I2C0),
//                                   (volatile uint32_t)(AR_IO_CREG_MST0_CTRL));

      /* disable device */
      WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(AR_IO_I2C_MST0_CON+I2C_CON)) & ~(I2C_ENABLE_MASTER),
                                   (volatile uint32_t)(AR_IO_I2C_MST0_CON+I2C_CON));

      interrupt_unlock(saved);
      }
}

void setup() {
  // put your setup code here, to run once:
    Serial.begin(115200);
    while(!Serial);

    adc_normal_wo_calib();

    creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);

    Serial.print("Master creg init: ");
    Serial.print(creg, BIN);
    Serial.println();  

    creg = READ_ARC_REG(AR_IO_I2C_MST0_CON+I2C_CON);

    Serial.print("AR_IO_I2C_MST0_CON+I2C_CON init: ");
    Serial.print(creg, BIN);
    Serial.println(); 

    disableI2C0();

    creg = READ_ARC_REG(AR_IO_CREG_MST0_CTRL);

    Serial.print("Master creg after disabling I2C0: ");
    Serial.print(creg, BIN);
    Serial.println();  

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Output:

> Master creg init: 10000000000000000000000000000100
> AR_IO_I2C_MST0_CON+I2C_CON init: 0
> Inside disableI2C0() AR_IO_CREG_MST0_CTRL: 10000000000000000000000000000100
> Inside disableI2C0() AR_IO_I2C_MST0_CON: 0
> creg after I2C control bit set: 10100000000000000000000000000100
> After write to AR_IO_CREG_MST0_CTRL: 10100000000000000000000000000100
> Master creg after disabling I2C0: 10100000000000000000000000000100

eriknyquist, 2016-07-25:
Thanks @claudiaAl, we'll take a look

eriknyquist, 2016-07-26:
@claudiaAl I'm still not really sure what you are asking. I can run the code, but it is unclear to me what the issue is. Can you explain why your output is not what you expected? what did you expect to see?

claudiaAl, 2016-07-26:
@eriknyquist I'm confused about what to set the the bit to that disables/gates the I2C0 and I2C1 clocks.  For reference I looked at the ss_i2s_iface driver code.  This is the function there used to disable the clock.

```
DRIVER_API_RC ss_i2c_clock_disable(I2C_CONTROLLER controller_id)
{
    i2c_info_pt dev = &i2c_master_devs[SS_CTRL_ID(controller_id)];
    uint32_t saved;

    /* Protect registers using lock and unlockl of interruptions */
    saved = interrupt_lock();

    /* disable clock to peripheral */
    WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(AR_IO_CREG_MST0_CTRL)) & ~( 1 << dev->creg_i2c_clk_ctrl),
                                   (volatile uint32_t)(AR_IO_CREG_MST0_CTRL));

    /* disable device */
    WRITE_ARC_REG(READ_ARC_REG((volatile uint32_t)(dev->reg_base+I2C_CON)) & ~(I2C_ENABLE_MASTER),
                                   (volatile uint32_t)(dev->reg_base+I2C_CON));

    interrupt_unlock(saved);

    return DRV_RC_OK;

}
```

If you look at the first `WRITE_ARC_REG` function that disables the clock to the peripheral, the bit that is written to the `dev->creg_i2c_clk_ctrl` bit in the CREG master control register always is a 0. 

`(AR_IO_CREG_MST0_CTRL)) & ~( 1 << dev->creg_i2c_clk_ctrl)` always evaluates to 0.

 But the EDS documentation states that the bit should be set to a 1 for the clock to be gated.  

> 
> **2.1.2.11 CREG master**
> [29]
> CLK_CTRL – I2C 0 clock gate – this is used to gate the clock to the I2C 0 Master
> [30]
> CLK_CTRL – I2C 1 clock gate – this is used to gate the clock to the I2C 1 Master
> 
> **2.1.6.86 CREG register detailed description
> 2.1.6.87 IO_CREG_MST0_CTRL (IO_CREG_MST0_CTRL) CREG control register**
> [30]  I2C1_CLK_GATE (I2C1_CLK_GATE)
> When I2C1_CLK_GATE is 1 it indicates that the I2C 1 peripheral is inactive and its clock can be gated.
> [29]  I2C0_CLK_GATE (I2C0_CLK_GATE)
> When I2C0_CLK_GATE is 1 it indicates that the I2C 0 peripheral is inactive and its clock can be gated.

So, my questions are:
1.  Is the driver code correct in that setting the I2C0 control bit in the CREG master control to '0' gates the clock?
2.  Ultimately, to disable the I2C buses, should I be setting the control bits to 1 or 0?

I hope that makes better sense! Thanks :-)

eriknyquist, 2016-07-26:
I'm not sure, to be honest. If the documentation + code is ambiguous, then I guess the easiest way to find out would be to just test it.

1. Shift some bytes out of I2C0
2. Verify the bytes on the wire with a scope / logic analyser
3. Set the control bit to 0
4. Try to shift more bytes out on I2C0
5. If nothing comes out, then you know I2C0 is disabled

claudiaAl, 2016-07-28:
I brought out the scope and I discovered an issue with my code.  Setting my I2C clock bit only works if the CREG Master is not empty.  I have posted my issue on stack overflow and have tried all of the suggestions but nothing seems to work.  I'm not sure what I am missing though...
http://stackoverflow.com/questions/38643407/bit-setting-logic-in-c

claudiaAl, 2016-07-28:
I'm working through some issues in my code to get this up and running and I am not convinced that I am reading a register correctly.  I am wanting to look at the I2C0 and I2C1 Control registers to make sure that they are disabled.  However, I am not seeing a change at all regardless of setting or clearing the bit.  
```

WRITE_ARC_REG(READ_ARC_REG((AR_IO_I2C_MST0_CON+I2C_CON)) & ~(I2C_ENABLE_MASTER),
                                   (AR_IO_I2C_MST0_CON+I2C_CON));
creg = READ_ARC_REG(AR_IO_I2C_MST0_CON+I2C_CON);

Serial.print("AR_IO_I2C_MST0_CON+I2C_CON after: ");
Serial.print(creg, BIN);
Serial.println(); 
```
When the I2C is enabled the bit should be set to a 1.  In both cases (enabling and disabling) my printout of `AR_IO_I2C_MST0_CON+I2C_CON` is always 0.  Am I reading the register incorrectly?

claudiaAl, 2016-07-29:
I am reading it correctly.  My writes are not working as intended but I think that we can call this closed.  Thank you!!

closed 2016-07-29 by claudiaAl