arduino/ArduinoCore-arc32 #254: GATT Client API needed

bluetooth-mdw, 2016-07-29:
It would be great if there was a class which provides a GATT client API. Some profiles have the GAP Peripheral as the GATT Client (e.g. the HTTP Proxy Service) so that once a connection has been accepted by the peripheral it then acts as GATT client and sends ATT PDUs over the connection to the GATT server at the other end which was previously the GAP Central device.

Without this, it's not possible to use Arduino 101 in IoT scenarios where the peripheral formulates and instructs a gateway device to execute HTTP requests on its behalf... the HTTP Proxy Service is a Bluetooth SIG standard designed for this purpose. It's not the only time you might want a GAP Peripheral to be the GATT Client but it's a key example. GATT and GAP roles are not coupled in any way.

The master branch of the original arduino-BLEPeripheral does seem to have this capability (not verified).

See

https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/73

and

https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/master/examples/remote_test

eriknyquist, 2016-08-01:
@SidLeung can you review? not sure if  this is related to https://github.com/01org/corelibs-arduino101/pull/255

SidLeung, 2016-08-11:
Will investigate where our BLE library diverged from Sandeep.

eriknyquist, 2016-10-10:
@SidLeung any updates on this?

noelpaz, 2016-10-11:
No updates on this. Not sure how far along are we are on the refactor.

SidLeung, 2016-10-31:
The upcoming release BLE library allows the creation of Service for both Central and Peripheral mode.  In either mode, the board can be a server or a client.

kitsunami, 2017-02-21:
Hi @bluetooth-mdw, can you retest with the latest, found her: http://forum.arduino.cc/index.php?topic=443728.0 ?

bluetooth-mdw, 2017-02-27:
I will but it might be a while before I can schedule this. Thanks for the update.

closed 2017-03-13 by kitsunami