arduino/ArduinoCore-arc32 #256: I2C Disable Command Flow

claudiaAl, 2016-08-01:
Hello!  I am working on a low power library and am currently working on disabling the I2C0 and I2C1 interfaces. I have two issues that I am trying to tackle.  

#1)  I have been able to verify that I can read and manipulate the registers that I need to in order to accomplish this.  I am a little confused on the flow of what needs to happen though.  I referenced the i2c interface driver code, but am still unclear.  Here is some quick pseudocode.. can anyone confirm that this is correct, or perhaps steer me in the right direction?  

> lock interruptions
> disable the clock to the peripheral in CREG Master
> disable the clock in the I2C0 Master Control register (DOES THIS NEED TO HAPPEN?)
> disable the device in I2C0 Master Control register
> unlock

#2)  Per @eriknyquist suggestion, I have an analyzer hooked up to the data and clock lines used by what I think is I2C1 so that I can figure out whether the clock enable bit is supposed to be a 1 or a 0 (documentation and code seem to be inconsistent).  I can see that loading a completely empty sketch actually initializes the clock line for some reason and I have absolutely no idea why.  I was hoping to test something like:

```
Wire.begin();
Wire.beginTransmission(8);
Wire.write("Data to write");
Wire.beginTransmission(8);
Wire.endTransmission();
disableI2C1();
Wire.write("Data to write");
Wire.beginTransmission(8);
Wire.endTransmission();

```
Where I wouldn't see the clock line on my analyzer for the second write... maybe?  Can anyone think of a better way of testing whether or not my I2C clock is indeed gated and if the interface is disabled?

Thanks in advance!
-c

eriknyquist, 2016-08-02:
1) I'm not sure what you mean by "does this need to happen", in reference to disabling I2C clocking. If you are wondering whether disabling the I2C clocking will result in less power consumption, well probably yes. Test it and see.

2) What do you mean by "loading en empty sketch initialises the clock"? Do you see a consistent, repeating, clock signal? or is it just some noise on the line? Does it happen when you call Wire.begin(), when the sketch is being uploaded, or what? Need a little more info here, and perhaps a photo of the signal capture if you have it.

claudiaAl, 2016-08-04:
1) I was wondering if there is a best practice for disabling devices.  A coworker suggested I first make sure all (if any) processing is finished, disable the clock, then disable the device.  He said this would gracefully disable the device in the event that it is in use.  

2) I'm still fighting with the I2C code-- I'll update as soon as I have something more substantial.

eriknyquist, 2016-08-05:
Well, there's really no best practice because every peripheral device is different. So,  I guess you could say the best practice would be to "read the hardware spec (which you now have) and do what it tells you".

Additionally, I would add that I don't think it's the responsibility of your library to check if something is in use before disabling it; your library should do exactly what it says on the tin- disable devices to reduce power consumption. IMO it's the responsibility of the user to ensure that, for example, Serial is not in use before disabling it (and we even have Serial.flush() and Serial.end() for that).

claudiaAl, 2016-08-12:
So I have been able to confirm that the Curie EDS documentation is incorrect for how some bits get set in the CREG Master register.  Specifically, when disabling the I2C interfaces, the clock gate bit in CREG Master needs to be cleared.  Documentation states that it needs to be set to a '1'.  

We ran a vey simple test that had a digital temperature sensor sending data over the I2C bus.  We went into i2c.c and modified the `i2c_openadapter()` function.  Rather than calling `ss_i2c_clock_enable()` we called `ss_i2c_clock_disable()` to see if we would still receive data.  We didn't which means that the code is correct, but the doc is not.  Section 2.1.6.86 CREG register detailed description shows that the bits should be set rather than being cleared...

eriknyquist, 2016-10-24:
closing this, since no activity for a couple of months, and @claudiaAl knows now that this EDS document is old and she shouldn't use it anymore.

closed 2016-10-24 by eriknyquist