arduino/ArduinoCore-arc32 #258: New BLE Central Preview Code -- BLE Peripheral is not advertising

noelpaz, 2016-08-02:
I am using this JSON http://mkfs.ndg.intel.com/precommit/json/package_0.0.41_index.json and this flashpack http://mkfs.ndg.intel.com/calvin/flashpack_ble_v3.tbz2  (note to non-Intel members you will not have access at this time). @calvinatintel can comment on the correct commit SHA for reference.

I have seen 2 major issues so far as a user doing some preliminary tests.

PIN 13 LED is disabled meaning the basic test Blink  if uploaded, the sketch will not work as expected. If you go back to the release version 1.0..6 Blink works again

As a peripheral using the new corelibs and the LEDCallback Example -- the device is now NOT scannable as BLE peripheral. I have tried 2 phones and BlueZ hcitool which is part of bluez and I cannot see it. If you go to the 1.0.6 corelibs using the same example sketch you will get a compile error

If you use the 1.0.6 example using the 1.0.6 corelibs with the new BLE FW it also does not work unlike Blink. It seems to me that corelibs and maybe the changes in BLE FW may need some investigation.

Thanks

Noel

eriknyquist, 2016-08-02:
@noelpaz I tried the Blink sketch, the Scanning (central) sketch and the BatteryMonitor (peripheral) sketch, using the latest V3 Nordic firmware, and they all worked fine. I am using the Nordic Android app, called "NRF connect".

noelpaz, 2016-08-03:
That is also what I used. But I used the LEDCallback as that exercise reading and writing characteristics. Let me see if BatteryMonitor works

noelpaz, 2016-08-03:
@eriknyquist 

I tried that. I made sure I used V3

[image]

Then I made sure it is the correct corelibs  0.0.41  -- heres an output during compilation
"C:\Users\nfpaz\AppData\Local\Arduino15\packages\Intel-Test\tools\arc-elf32\1.6.4+1.0/bin/arc-elf32-g++" .... IC:\Users\nfpaz\AppData\Local\Arduino15\packages\Intel-Test\hardware\arc32\0.0.41\system/libarc32_arduino101/common" "-IC:\Users\nfpaz\AppData\Local\Arduino15\packages\Intel-Test\hardware\arc32\0.0.41\system/libarc32_arduino101/drivers" "-....

Yes the Battery Monitor Sketch works but the LEDCallback example does not and the difference I see is that the Battery Service is known service with UUID 180F  but the LEDCallback is a unique user provided UUID

//BLEService ledService("19B10000-E8F2-537E-4F6C-D104768A1214"); // create service
BLEService ledService("180F");

If I do this I could connect however the Centrals address is not correct as it reports it as 00:00:00:00:00. You cannot also see the characteristics of the custom service
The way they handle connection is also different. The Battery Monitor is just waiting for the central object while  the CallbackLED is calling the poll function  (blePeripheral.poll()) and has a callback function to indicate connection and disconnects.

Anyway please try the CallbackLED ftrom 0.0.41 and let me know if you get the same results

eriknyquist, 2016-08-03:
Yes, the CallbackLED example does not work for me either

noelpaz, 2016-08-03:
I assigned the internal JIRA to Liang. I cannot find the member name here.

noelpaz, 2016-08-03:
I found the issue. The UUID on the ledService in the example is too long. It cannot be more than 128 bits or 31 characters + NULL. I looked at BLEPeripheral.h and the corresponding .cpp file

In the sketch CallbackLED.ino

BLEService ledService("19B10000-E8F2-537E-4F6C-D104768A1214");
If you used this you should be good
BLEService ledService(("19B10000-537E-4F6C-D104768A1214")

In the past it supported longer than 31.

This would affect setting the advertised UUID

    /**
     * Set the Service Data that the BLE Peripheral Device advertises
     *
     * @param serviceDataUuid 16-bit Service UUID for this Service Data
     *   (in string form). Must match the UUID parameter
     *   of setAdvertisedServiceUuid(). To fit into BLE_MAX_ADV_SIZE,
     *   the UUID must be a 16-bit UUID.
     *
     * @param serviceData binary array of Service Data.
     *
     * @param serviceDataLength length (bytes) of serviceData[]
     *
     * @note the entire advertising packet must be no more than
     *   BLE_MAX_ADV_SIZE bytes, which is currently 31.
     *   This likely means that if you use Service Data
     *   there will not be room for a Local Name.
     *
     * @note if serviceDataUuid isn't 16-bits long, or if
     *   serviceDataLength won't fit in the advertising block,
     *   the service data will silently not be copied
     *   into the advertising block.
     */
    void setAdvertisedServiceData(const struct bt_uuid* serviceDataUuid, 
                                  uint8_t* serviceData, 
                                  uint8_t serviceDataLength);

eriknyquist, 2016-08-03:
nice one @noelpaz ... can you provide a fixed example sketch that I could test?

eriknyquist, 2016-08-03:
Oops sorry.... you already provided a shorter UUID that works. OK cool I'll try it.

noelpaz, 2016-08-03:
However the central's BT address is incorrect and defaulting to 00:00:00:00:00:00. Something is probably up with the poll() method. In the Battery monitor it returns the centrals address.

eriknyquist, 2016-08-03:
I tried using the shorter UUID, and it still didn't work-- I can't see the bluetooth device advertising

sgbihu, 2016-08-04:
I have found the root cause. The reason is I disable some debug trace. The UART have a high load when register the profile. The BLE Lib set the ADV data after register the profile. It will makes UART overload. And the ADV packet data is lost. And the sniffer can capture the device's ADV. But no ADV data. 
In the release version all debug trace will be disabled. I test with the debug version. I also find this issue when I push to the Github. But I disabled some debug trace when I change code based on code review. That make issue happen again. Because the UART have a limitation.

sgbihu, 2016-08-04:
I have tested the latest version on my branch. And it work without debug trace. So you can test it now. 

If you find the basic function doesn't works, you can report bug and enable the debug trace on Serial1 (The steps can be found in the README.md). And told me the issue's state.

eriknyquist, 2016-08-09:
@noelpaz can this issue be closed?

noelpaz, 2016-08-09:
Hi It got fixed but when they made a change on Tuesday with this SHA 6d81321

This issue resurfaced again

noelpaz, 2016-08-09:
It looks like the issue is related to the BLEDescriptor. the user has to specify a size for the descriptor value so ignore the comment below. Please see internal JIRA for details

noelpaz, 2016-08-09:
I think we can close this issue.

closed 2016-08-09 by noelpaz