arduino/ArduinoCore-arc32 #265: Constant values to atan2 dont result in constant output

epic11, 2016-08-11:
I need to use the built in atan2 function, but even with constant input it sometimes gives me completely wrong values. Example from my Serial Monitor output:
1.51
1.51
1.51
1.51
1.51
1.51
86050.79
1.51
1.51
1.51
1.51
1.51
1.51
1.51

By printing:
Serial.println(atan2(454.00, 17206.00) * RAD_TO_DEG);

PaulStoffregen, 2016-08-11:
Can you give us more info?  Think "reproducible", everything someone would need to recreate this problem.  Right now, we don't even know which board you're using, not to mention having a complete program to copy & paste into Arduino and upload to that board.

epic11, 2016-08-12:
Sorry, for leaving such a short issue. I realized that this bug has something to do with handling interrupts in another part of my code, when I disable it atan works again, so it has probably something to do with my own code, I was just really confused by such a simple function not working.

closed 2016-10-24 by kitsunami