arduino/ArduinoCore-arc32 #267: Function behavior differs based on caller

claudiaAl, 2016-08-12:
Hello!  

I have acquired a bench power supply and have started to play with measuring my current to identify power savings.  I have my multimeter in series at the Curie's VSYS.  I am really confused by the behavior of something though.  

In variant.cpp, three clocks are enabled when a sketch is first loaded (GPIO, PWM, and ADC).  If I comment out the `variantGpioInit()` and `variantPwmInit()` functions, leave the variantAdcInit() function, and upload an empty sketch, my current measurement is at around 29.53-29.68 mA.

If I comment out all three init functions in variant.cpp, but copy the `variantAdcInit()` function into my sketch in the `setup()` function, my current reading drops down to 29.04-29.18 mA.  

This makes me wonder if different processors are handling the initialization that happens in variant.cpp and what goes on in the sketch... can someone shine a little light on this for me? 

Thanks!

eriknyquist, 2016-09-09:
@claudiaAl don't really understand the question. Are you asking if your sketch and the code in variant.cpp are executed by different processors? (if so, the answer is no- they are both executed by the ARC core).

claudiaAl, 2016-09-14:
Yes, that was the question.  I'm not really sure why there was a difference, thank you for the clarification.

closed 2016-09-14 by claudiaAl