arduino/ArduinoCore-arc32 #268: Public interface for Device's MAC address

ccoulton, 2016-08-15:
I'm working with the Arduino 101 board to create a project, and would like to access the ble_addr_t or _local_bda, to generate an EUI64 for the device. https://standards.ieee.org/develop/regauth/tut/eui64.pdf

This Global identifier requires the MAC address of the device, with a 0xff or 0xfe byte inserted in the middle of the 48 bit address of the MAC, Being able to do this programmatically so each device could determine it's address w/o having to change the sketch for each device would be great!

ccoulton, 2016-08-16:
I was able to do this by getting a ble_addr_t from the ble_client_get_factory_config method, but something forward facing would be nice

eriknyquist, 2016-10-10:
@SidLeung can you comment? is it possible to provide a public interface for retrieving the MAC address?

noelpaz, 2016-10-11:
If you look at the current CurieBLE and look at Scanning example we are displaying the Mac Address of the BLE peripherals being scanned. The peripheral can also get the central's mac and also any central app had access to the peripheral's MAC address.

But you want to know the mac address without the help of a central or another peripheral. Basically a self-discovery, correct

I looked at the include files and we do not have that implemented.  But we are refactoring to @sandeepmistry BLE library , if that is implemented there then it maybe available as well

https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/src/BLEUtil.h  . the API 
static void addressToString(const unsigned char *in, char* out)   looks just like a utility to convert a known Mac Address to the xx:xx:... form.

sandeepmistry, 2016-10-11:
The new API we've been discussing will have a `BLE.address()` API that returns a String for the local BT address.

kitsunami, 2017-02-21:
Can you please retest this on the latest release found her: http://forum.arduino.cc/index.php?topic=443728.0 , @ccoulton

ccoulton, 2017-02-24:
Will do when I make it back into the office

closed 2017-03-13 by kitsunami