arduino/ArduinoCore-arc32 #270: [DOC] Reference/CurieIMUreadTemperature Wrong formula

GregorioW, 2016-08-22:
I’m aware that the following is not a crucial issue, but I hope it’s not a big deal to correct it (and that it’s the right place to report it).

In the [CurieIMU.readTemperature() reference](https://www.arduino.cc/en/Reference/CurieIMUreadTemperature) the conversion example is wrong.
Instead of
Celsius=(raw/32767.0)+23
it should be
`tempCelsius = rawTemperature / 512.0 + 23.0;`
(see [BMI60.cpp:2272](https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieIMU/src/BMI160.cpp#L2272))

kitsunami, 2017-02-21:
@facchinm can you take a look at this and update the webpage if needed?

facchinm, 2017-02-22:
Thanks for reporting, fixed :slightly_smiling_face:

closed 2017-02-24 by kitsunami