arduino/ArduinoCore-arc32 #272: Wrong register bits referenced in doc (EAS and EDS)

claudiaAl, 2016-08-23:
The CREG Master 0 Control Register has 5 bits (bits 27-31) that control the clock gating of peripherals (i.e. ADC, SPI0, SPI1, I2C0, I2C1) .  The EAS doc (pg 71) and the EDS doc (pg 228) both list these incorrectly as bits 24-28.

kitsunami, 2016-09-20:
Tracking fix via /CURIE-60

claudiaAl, 2016-10-21:
Hey @eriknyquist , my apologies with posting about the EAS doc, I didn't realize it was internal only.  The Curie Module EDS version that I had been using was Rev 0.52 and I received it from a peer.  I'm not sure if there is an updated version for this available or not... a quick search didn't bring anything up for me but I could just be missing it.

eriknyquist, 2016-10-21:
OK. Do you have a link to the public version of this document? I don't see anything called EDS in the public docs for the Curie module, or Quark SE.

claudiaAl, 2016-10-21:
Hey Erik, I asked the manager that provided me with my copy for the original location of the file and he thought that it came from iRef.  I did a search and was only able to find the data sheet and platform design guide.  I've attached a copy for your reference.. not sure if it is out there anywhere!

[curie-EDS_559322_0-52 copy.pdf](https://github.com/01org/corelibs-arduino101/files/545430/curie-EDS_559322_0-52.copy.pdf)

eriknyquist, 2016-10-21:
Closing,couldn't find this discrepancy in any of the currently published documents

closed 2016-10-21 by eriknyquist