arduino/ArduinoCore-arc32 #280: digitalWrite and analogWrite do not work in combination

khanning, 2016-09-02:
If you call analogWrite on a PWM capable pin with any value other than 0 and 255, any subsequent  calls to digitalWrite on the same pin are ignored until one of the following events occurs:

1. ```pinMode(PIN, OUTPUT)``` is called on the pin again
2. You call ```analogWrite(PIN, 0)``` or ```analogWrite(PIN, 255)```

This behavior does not occur on the Arduino UNO. 

Here is a simple sketch that illustrates the issue:
```
void setup() {
  pinMode(9, OUTPUT);
}

void loop() {
  digitalWrite(9, HIGH);
  delay(1000);
  analogWrite(9, 255/2);
  delay(1000);
  digitalWrite(9, LOW);
  delay(1000);
}
```

**On the 101**
Once ```analogWrite(9, 255/2)``` is called, pin 9 remains at a 50% duty cycle, ignoring all subsequent digitalWrite commands.

**On the UNO**
Pin 9 successfully switches between 100%, 50%, and 0% duty cycle.

eriknyquist, 2016-09-29:
@bigdinotech can you look into this please?

bigdinotech, 2016-10-05:
That is actually the expected behavior.
when using analogWrite() the pins are muxed into pwm mode
when using pinMode() the pins are muzxed into gpio mode
We are not muxing during digitalWrite() for performance reasons.
analogWrite(pin, 0) and analogWrite(pin, 255) works because we are actually switching to gpio modes for those values

facchinm, 2016-10-25:
@bigdinotech I believe that using a bitmap (in RAM) to keep the pin mux state would solve the issue. Since you got much improvement in RAM usage since 1.0.7 we could use some of this spare memory to avoid this kind of problems (which are very time consuming for people porting AVR libraries). For example, we did this on SAM core https://github.com/arduino/Arduino/commit/d1c65ca8fe4fc0a81f6760cc9abcaf140113e214 
Benchmarking the solutions would help also (checking the actual mux should take like a couple of instructions if the array is well crafted)

eriknyquist, 2016-10-25:
Re-opening since @facchinm's last comment sounds like something we should do

eriknyquist, 2016-12-01:
Once #355 has been verfied and merged, this issue can be closed.

closed 2016-12-19 by bigdinotech