arduino/ArduinoCore-arc32 #300: [Core 1.0.7] Print floats regression

facchinm, 2016-09-27:
This commit https://github.com/01org/corelibs-arduino101/commit/3168ede2cbd3df99121944d46a5b24e9557d2b4b causes some floats to print incorrectly. Reverting is not a chance so it should be investigated.

The original report is here http://forum.arduino.cc/index.php?topic=426292.0, example sketch attached 
```
// Please comment out the below line if you are using an Arduino Uno
#define M_PI 3.14

// unsigned long to store time because ms gets large quickly
unsigned long previousTime = 0;

// constant of time (milliseconds) to update
const long interval = 10;

// omega of the sine function in radians/s
// M_PI is pi = 3.1416 in Arduino code
float omega = M_PI;

// sine function's angle
float angle = 0.0;

void setup() {
  // make sure Serial Plotter is open to visualize data in graph form
  Serial.begin(9600);
  delay(2000); // wait until the board fully wakes up.
}

void loop() {
  // find current time (milliseconds)
  unsigned long currentTime = millis();
  // if time difference is greater than the update interval, then
  // send next angle to Serial Plotter
  if (currentTime - previousTime >= interval) {
    previousTime = currentTime;
    // update angle incrementally
    angle = angle + ((float)interval / 1000.0 * omega);
    // when phase passes 2*pi reset to 0 so phase stays between 0 & 2*pi
    if (angle >= 2.0 * M_PI) {
      angle = angle - 2.0 * M_PI;
    }
    float sineAngle = sin(angle);

    Serial.println(sineAngle);

  }

}
```

eriknyquist, 2016-09-27:
Thanks for the heads up @facchinm, I'll look into it.

eriknyquist, 2016-09-28:
It was a faulty implementation-- I missed a corner case. @facchinm please test https://github.com/01org/corelibs-arduino101/pull/302

eriknyquist, 2016-09-29:
Merged https://github.com/01org/corelibs-arduino101/commit/2e56b81a264c8760f095a12f650381b8ea8de831

closed 2016-09-29 by eriknyquist