arduino/ArduinoCore-arc32 #309: Arduino 101 Core 1.0.7 cannot compile!!

ralphxyz, 2016-09-30:
Windows 10 x64 Arduino 101

the error:

C:\Users\Ralph>cmd /C copy /y "C:\Users\Ralph\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp\Blink.ino.elf" "C:\Users\Ralph\AppData\Local\Temp\build9c9ef3bdfe2fccb480bc6e4bac749e41.tmp\..\arduino101_sketch.debug.elf"
'cmd' is not recognized as an internal or external command,
operable program or batch file.

The compiler user cannot run "cmd" only administrator can run cmd!!

Version 1.0.7 and Version 1.0.6 both fail!!

Version 1.0.5 compiles successfully!!

Version 1.0.5 does not select a "programmer" from Tools.

eriknyquist, 2016-09-30:
Thanks for letting us know @ralphxyz, we'll try to reproduce these errors and get back to you soon.

ralphxyz, 2016-10-01:
A workaround solution for this problem would be to run the "Programmer" as Administrator,
might have to ask permission.

A User cannot run the cmd command only a Administrator can run cmd!!

This is possible/probable not a bug in Version 1.0.7 but a configuration
of my Windows x64 computer.

It might be Windows or even my Norton Security protecting me from my self.

Version 1.0.7a would run the Programmer as Administrator.

I am trying to get Raspberry Pi running so I can try using Linux.

Ralph

facchinm, 2016-10-03:
For reference, there is [this](http://forum.arduino.cc/index.php?topic=425387.msg2942374#msg2942374) post on the forum reporting the same problem.
@eriknyquist until not needed by the end user, I would comment out those lines on the released core in order to avoid this kind of problems

noelpaz, 2016-10-04:
Hi. I am a colleague of @eriknyquist  and tried to replicate this problem today. After reading what is here I think we are dealing with problems compiling and uploading (let us bundle this as one operation) and running the command burn bootloader.  The other issue also seems to be that the user has non administrative privileges.

Here is my setup Windows 10 professional 64 bit, Arduino IDE 1.6.12 and Arduino 101 core libraries and tools version 1.0.7. Windows Defender is the various scanning software. The latest updates recommended by Microsoft has been installed.

I also disabled the ability of the user running the command line. Steps to that are here https://www.maketecheasier.com/disable-command-prompt-windows/

Now there are 2 options to this setting. You can just disable the command line console or you can disable command script processing.

If I do this then I can replicate the issue. My suggestion is to reverse this setting if you can (you have to be an admin). 

But by default a non admin user can upload, run the programmer to update firmware and download a sketch. Please verify this setting by following the link above.

ralphxyz, 2016-10-04:
I am an Administrator.

The problem is that the programmer runs as a user not an Administrator.

I do not use Windows Defender, I use Norton Security, I am not sure there
is a override option in Norton.

facchinm <https://github.com/facchinm>'s work around eliminated the cmd
problem.

Now I am getting a different error.

noelpaz, 2016-10-04:
[image]

This is a screenshot of the dialog from gpedit (global policy editor) . These are the settings which replicates the problem for the non-admin user. Reversing this would eliminate the problem. 

Assigning this to @calvinatintel  so that he and @facchinm can discuss the [workaround in the Arduino 101 forum](http://forum.arduino.cc/index.php?topic=425387.0)

noelpaz, 2016-10-04:
@ralphxyz -- Yes I can confirm that with the policy setting set so that Command prompt is disabled and command script processing also disabled, even as an administrator you cannot compile or upload for Arduino 101 board but can with other Arduino boards.

The thing is that even though you are administrator, you are running most programs as a regular user. That is why if you create a folder in Program Files you will get a UAC* dialog asking if you want to proceed. Also some programs are written defensively, give only access needed. The argument here is that if the system administrator wishes to enable some security measures then the programs should follow suit.   I am sure that @calvinatintel and @facchinm can some to an agreement on how to resolve this.

Please check the settings as described in the link above. If your virus scanner is doing some other protection strategy beyond built-in tools by microsoft, then for now, you have to only implement the work around by @facchinm  on similarly configured on affected systems.

Thanks

*UAC User Access Control

calvinatintel, 2016-10-04:
Hello @ralphxyz,

Did you try this? http://stackoverflow.com/questions/32035577/cmd-is-not-recognized-as-an-internal-or-external-command-operable-program-or

noelpaz, 2016-10-04:
It was pointed out to me by @calvinatintel that the original error message could be a symptom that cmd is not in the user's path.  To replicate:

1. I created an admin user
2. I opened a command line and then I type set path. I copied my path in a notepad and removed C:\Windows\System32  where usually cmd.exe is located
 ```
C:\Users\ndgtester> Set Path=C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\
```
3. Then I star Arduino from within this command line ..  cd to my installation directory and just type arduino.exe
4. Compiling WITHOUT Martino's workaround gives this error. 
```
cmd /C copy /y "C:\Users\NDGTES~1\AppData\Local\Temp\arduino_build_319682\sketch_oct04a.ino.elf" "C:\Users\NDGTES~1\AppData\Local\Temp\arduino_build_319682\..\arduino101_sketch.debug.elf"
exec: "cmd": executable file not found in %PATH%
```
@ralphxyz . If cmd is not in your path, then something must have removed c:\Windows\system32 from your path or cmd.exe has been removed.

Anyhow @facchinm workaround handles both root causes.

facchinm, 2016-10-05:
Removing completely those line in "production" is the way to go in my opinion. They are unused and the IDE is still lacking an integrated debugger to take care of this. The problem here is that `strip` command runs with only one parameter, so the original file with debugging symbol gets overridden. Once the IDE acquires the debugger capability, we'll surely handle this case since we need the unstripped file.

calvinatintel, 2016-10-06:
@facchinm Replacing `cmd /c copy ...` with `%COMSPEC% /c copy ...` failed as you thought because IDE couldn't expand the environment variable.
```
%COMSPEC% /c copy /y "C:\Users\scpark\AppData\Local\Temp\arduino_build_432450\sketch_oct05a.ino.elf" "C:\Users\scpark\AppData\Local\Temp\arduino_build_432450\..\arduino101_sketch.debug.elf"
exec: "%COMSPEC%": executable file not found in %PATH%
```
I wanted to run `set PATH=%COMSPEC%;%PATH%` before running the `cmd /c copy ...`, but that wasn't possible because `set` is a subcommand of `cmd`. I can't think of a way to modify that line to make it work for someone who has removed `c:\windows\system32` from `%PATH%`.

I can remove the line from `platform.txt`, but we do ship OpenOCD as a part of board package for debugging. Since the the unstripped `.debug.elf` is necessary for debugging wit OpenOCD, I'll just comment it out for now so that advanced users can uncomment it as needed.

Does that sound okay?

facchinm, 2016-10-06:
It sounds great! Commenting it is surely the way to go, so advanced user can easily get it working without searching over the web for the correct lines

ralphxyz, 2016-10-06:
First thing I have Windows32 in my  Path!! It was not there but that was
one of the first things I checked, it was not there so i added it and still
got the error!!

facchinm <https://github.com/facchinm>'s work around worked but then I got
a new error.

That has not been addressed.

Ralph

calvinatintel, 2016-10-06:
Hi @ralphxyz what is the new error?

ralphxyz, 2016-10-06:
oops I meant System32, that's the problem with doing things on the fly

ralphxyz, 2016-10-06:
Here is the [new error:](https://forum.arduino.cc/index.php?action=dlattach;topic=425387.0;attach=182624)

calvinatintel, 2016-10-06:
That looks like a compile error in using `NewliquidCrystal` library. Are you able to compile a blank sketch?

calvinatintel, 2016-10-06:
Fix in PR https://github.com/01org/corelibs-arduino101/pull/314

ralphxyz, 2016-10-06:
I get a error when compiling a blank sketch.

Strange NewliquidCrystal is the current recommended LCD library.

I only installed the library didn't touch anything.

Here is the latest blank sketch error:

Arduino: 1.6.11 (Windows 10), Board: "Arduino/Genuino 101"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Ralph\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Ralph\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Ralph\Documents\Arduino\libraries -fqbn=Intel:arc32:arduino_101 -ide-version=10611 -build-path C:\Users\Ralph\AppData\Local\Temp\build09939f8c19f726d2732168ac633a7748.tmp -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arc-elf32.path=C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.9+1.0.1 -prefs=runtime.tools.arduino101load.path=C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\arduino101load\1.6.9+1.28 -prefs=runtime.tools.flashpack.path=C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\flashpack\1.0.0 -prefs=runtime.tools.openocd.path=C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\openocd\0.9.0+0.1 -verbose C:\Users\Ralph\Documents\Arduino\HelloWorld_i2c\HelloWorld_i2c.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Ralph\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Ralph\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Ralph\Documents\Arduino\libraries -fqbn=Intel:arc32:arduino_101 -ide-version=10611 -build-path C:\Users\Ralph\AppData\Local\Temp\build09939f8c19f726d2732168ac633a7748.tmp -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arc-elf32.path=C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.9+1.0.1 -prefs=runtime.tools.arduino101load.path=C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\arduino101load\1.6.9+1.28 -prefs=runtime.tools.flashpack.path=C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\flashpack\1.0.0 -prefs=runtime.tools.openocd.path=C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\openocd\0.9.0+0.1 -verbose C:\Users\Ralph\Documents\Arduino\HelloWorld_i2c\HelloWorld_i2c.ino
Using board 'arduino_101' from platform in folder: C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7
Using core 'arduino' from platform in folder: C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7
Detecting libraries used...
"C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10611 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\system/libarc32_arduino101/common" "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\system/libarc32_arduino101/drivers" "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\system/libarc32_arduino101/bootcode" "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\system/libarc32_arduino101/framework/include"  "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\cores\arduino" "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\variants\arduino_101" "C:\Users\Ralph\AppData\Local\Temp\build09939f8c19f726d2732168ac633a7748.tmp\sketch\HelloWorld_i2c.ino.cpp" -o "nul"
"C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -D__ARDUINO_ARC__ -std=c++11 -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10611 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\system/libarc32_arduino101/common" "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\system/libarc32_arduino101/drivers" "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\system/libarc32_arduino101/bootcode" "-IC:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\system/libarc32_arduino101/frame [... cut ...] a"
C:\Users\Ralph\AppData\Local\Temp\build09939f8c19f726d2732168ac633a7748.tmp/core\core.a(main.cpp.o): In function `main':

C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\cores\arduino/main.cpp:43: undefined reference to `setup'

C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\cores\arduino/main.cpp:47: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino 101.

ralphxyz, 2016-10-06:
I have System32 in my path!!

ralphxyz, 2016-10-06:
Sorry Calvin I do not understand the "fix" I have System32 in my path.

calvinatintel, 2016-10-06:
PR https://github.com/01org/corelibs-arduino101/pull/314 removes the copy line which was causing problems.

It looks like you're compiling `HelloWorld_i2c.ino`. Can you try `BareMinimum` sketch that you can find in `File > Examples > 01.Basics > BareMinimum`?

eriknyquist, 2016-10-07:
**C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\cores\arduino/main.cpp:43: undefined reference to `setup'

C:\Users\Ralph\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.7\cores\arduino/main.cpp:47: undefined reference to `loop'**

When you say you can't compile a 'blank' sketch, it looks like your file is totally empty. You can't compile an empty file; you need to at least define the `setup()` and `loop()` functions, even if they are empty themselves.

ralphxyz, 2016-10-07:
Show you what I know.

Calvin had asked if i can compile a blank sketch, so that is what I tried.

I can compile BareMinimum!!

calvinatintel, 2016-10-07:
PR https://github.com/01org/corelibs-arduino101/pull/314 merged

closed 2016-10-07 by calvinatintel