arduino/ArduinoCore-arc32 #318: Bluetooth doesn't start unless device is reset

Tstek, 2016-10-12:
We've had a problem recently where the Arduino 101's BLE will never start advertising (and maybe the rest of the code won't work, it's very hard to tell) until the reset switch is pressed (either, I think, but we usually used the regular (not master) reset).  This seemed to only occur when powering the device by the barrel jack alone, it would always work if it was powered by only USB or when both are plugged in. 

Thinking it might be some sort of race condition, I tried delaying at the start of the code. This made the board never work. So I took the opposite approach, removing any extraneous lines before the peripherial.begin() call. This included all Serial.print lines, and Serial.begin (all only there for debugging purposes anyway). 

For some reason, this appears to work -  so basically I think the issue is some conflict between Serial and BLE, which might explain why it works fine if the device has a USB connection, maybe if the 101 can't find a usb, the Serial.Begin call hangs the Arduino?

bigdinotech, 2016-10-12:
@Tstek are you running one of the examples? If so which one?
Are you able to share the code that was working and the code that was not working for you?

Tstek, 2016-10-12:
@bigdinotech uh, I guess nevermind, thanks for asking though, I just went through and reverted all the changes, to make sure I could give you the minimal amount of code necessary....and I couldn't get it to fail anymore. It was more or less the BLE LED Callback example, but with a bunch of stuff added on top. Maybe some board update fixed our problem while I wasn't looking, and I just re-loaded the code today (it has been a couple weeks since I changed the code).

I guess I'll just close this then, thanks.

closed 2016-10-12 by Tstek