arduino/ArduinoCore-arc32 #320: Bluetooth Advertising does not stop when BLEPeripheral.end() called.

Prof62, 2016-10-13:
When calling BLEPeripheral.end() advertising should stop but it does not. 

Looking at the source the behaviour of end() app it appears that either the current central will be disconnected or advertising will stop, but not both and it always seems to choose disconnect, even when already disconnected.

As a workaround I have modified my local copy of the V1.0.7 CurieBLE  library to add a basic wrapper function to manually stop advertising. 

Because my simple implementation doesn't clear any statuses I have had to implement a matching  start advertising method because init() wouldn't always restart advertising. (Possibly issue #231 has a similar underlying cause)

As an aside, whilst aware of the dangers of feature creep, specific start and stop advertising functions might be more intuitive (and feel less brutal), than calling init() and end() to do it. To me at least, init means "I'm setting this ready to use for the first time" and end means "I'm completely done with this thing, please destroy it".

noelpaz, 2016-10-14:
@Prof62  - You are correct on your observation. We did fix these API calls on the current github version of corelibs but also we have added BLE Central feautures. You will not get this version using the IDE (1.0.7) and should be treated as in-Development software. You will have to rewrite your sketches a little bit. If you look at the number of sketches you will see there are significantly more and will probably satisfy some of the features you mentioned above specially more control on advertising and connectable properties.

Prof62, 2016-10-14:
Thanks. I've now installed the latest version as you suggest but can't test it because I can't find locate the download location for a compatible flashpack to update my Genuino 101 firmware. (I also had to open the example files using a text editor but I suspect that is just an artefact of the manual install process)

I apologise for posting what is really a forum type question but I'm posting it here because it will help anyone trying to use this workaround so it made sense to keep the information together.

eriknyquist, 2016-10-14:
@Prof62 if you installed 1.0.7, update the BLE firmware this way:

Tools -> Programmer -> Arduino/Genuino 101
Tools -> Burn Bootloader

Prof62, 2016-10-14:
Thanks for the suggestion Erik.

The problem is that I have installed the Core libraries from this github repository following the readme instructions so they are not really 1.0.7. They also do not contain the flashpack which is in the tools directory higher up the tree. 

Unfortunately, this means that Burn Bootloader just re-installs the incompatible version.

Is there something else I need to do to force a download of the matching flashpack.

bigdinotech, 2016-10-14:
If you are trying to just get stuff from GitHub then you will also need to grab the new tools
https://github.com/01org/intel-arduino-tools

Prof62, 2016-10-14:
Thanks. Sorry for the noob question but I'm a bit too good at finding the wrong version of things and less good at picking the right one.

I'm assuming that pulling tools and core library is the official way to get the latest development build.

kitsunami, 2016-10-31:
This should be fixed in the next release.

closed 2016-12-19 by kitsunami