arduino/ArduinoCore-arc32 #321: I can NOT upload bootloader from 1.0.6 to 1.0.7

ouki-wang, 2016-10-14:
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:363)
	at processing.app.Editor.lambda$handleBurnBootloader$42(Editor.java:2752)
	at java.lang.Thread.run(Thread.java:745)
Error while burning bootloader.

eriknyquist, 2016-10-14:
Did you try shutting down the IDE and opening it again?

ouki-wang, 2016-10-19:
@eriknyquist 
yes，i does not work after open IDE again。i think that bat script's compatibility is not good

calvinatintel, 2016-10-19:
Hi @ouki-wang, there were no firmware changes between 1.0.6 and 1.0.7. Why did you need to burn the bootloader?

ouki-wang, 2016-10-20:
@calvinatintel ble_core image.bin is updated。

calvinatintel, 2016-10-20:
BLE firmware shipped with 1.0.7 is the same as the factory version. The upcoming 2.0.0 release will change the BLE firmware, so we've shipped the factory version with 1.0.7 as a way to return to the original version. If you're seeing a message that you need to update the firmware, you can completely reset back to the factory version using the [Factory Recovery Flashpack](https://downloadcenter.intel.com/downloads/eula/25470/Arduino-101-software-package?httpDown=https%3A%2F%2Fdownloadmirror.intel.com%2F25470%2Feng%2Farduino101-factory_recovery-flashpack.tar.bz2) and try again. Please let us know how it went.

closed 2016-10-24 by calvinatintel