arduino/ArduinoCore-arc32 #322: Problem with running Bluetooth examples

aleksandar-zivkovic, 2016-10-18:
Hi everyone, im running ubuntu and my board is functional (ive loading the 1.0.7 library) for basic examples like the blinking LED.

I copied this github repo as the instructions stated, to try and get the bluetooth battery monitering example working, but something is problematic with the platform.txt file when I copy it over. 

Specifically, for one, the sketch uses about x3 times more storage space, even for the most simple programs (11% to 30% for the blinking LED). Also, while I dont get compilation errors for the bluetooth program, I cant actually load it to the board...

Has anyone encountered a similar issue?

bigdinotech, 2016-10-18:
Hi @az51911 
If you want to use the update repo for corelibs you will also need to update the tools
https://github.com/01org/intel-arduino-tools

The increased sketch size was due to this https://github.com/01org/corelibs-arduino101/commit/1a5e3f7b2d65cf2b81417f85bd24ca455bdaa1fc
The reason for this change was to increase the amount of SRAM avaialble to the user

az51911, 2016-10-18:
Thank you @bigdinotech , do you know of any more detailed instructions on how to update the tools besides those given on that github page?

Here they are: "
    Download go package from here or using your package manager
    cd into the root folder of this project
    execute

export GOPATH=$PWD
export GOBIN=.
go get -d
go build
"

However, when i cd into the go directory, and run go get -d, this is what I get:

~/Downloads/go$ go get -d
can't load package: package .: no buildable Go source files

noelpaz, 2016-10-18:
@az51911 . I am assuming you are suing Linux. You need to perform all of this in the intel-arduino-tools/arduino101load folder.

If you did not install go using apt-get you need to add <path where installed go>/bin to your path.
You need to also set GOROOT which is the <path where installed go>.

If you are using mac you need to set GOOS as well

If you have a proxy to get to the internet then you also have to set the http_proxy and https_proxy variable. I actually got your error before when I did not set that properly.

I hope that helps

eriknyquist, 2016-10-18:
@bigdinotech that commit shouldn't change the size of the final binary. It's just moving an array from SRAM to flash.

eriknyquist, 2016-10-18:
@az51911 the difference in size is just because there is a lot of code in the BLE library. When you include the BLE library you pull it all into your sketch.

noelpaz, 2016-10-18:
BLE sketches have always been larger.  30 - 40% is normal.  I just compared a simple LED script to something that involves complicated functions and there is really not that much more difference. The underlying library is big and get's built whether you are using  those features or not.

claudiaAl, 2016-10-18:
Hello!  I wasn't sure if I should start a new topic.. but I am attempting to install the new tools so that I can use the most up to date libraries and came across an issue when I try to run 'go get -d'.  My error is: "fatal: unable to access 'https://github.com/mattn/go-shellwords/': Failed to connect to github.com port 443: Timed out" (exit status 128).  @az51911 , did you run across the same issue?

Thanks!

az51911, 2016-10-18:
@noelpaz, i ran all the instructions in the folder and they finished, but when I tried to upload the sketch, I still got Starting download script...
!! BLE firmware version is not in sync with CurieBLE library !!.

I'm assuming then that they weren't added to the path correctly? I ran export GOROOT=home/aleksandar I installed go using sudo apt install golang-go, in home/aleksandar, but still nothing.

bigdinotech, 2016-10-18:
@az51911 
did you run the burn bootloader step that updates the ble firmware?

az51911, 2016-10-18:
Yes @bigdinotech , I receieved the message: Done!
Resetting USB to switch back to runtime mode!!!SUCCESS!!!

But then when i switched back to the original programmer, I still received the same error.

eriknyquist, 2016-10-18:
what do you mean "switch back to the original programmer?"

You should be doing this;

Select "Tools->Programmer->Arduino/Genuino101"
Select "Tools->Burn Bootloader"

Then when it's done, you can try uploading sketches. No switching required.
Is that what you did?

az51911, 2016-10-18:
I meant the AVRISP mkll, I tried both and neither resulted in a successful uploaded sketch.

eriknyquist, 2016-10-18:
AVRISP mkll is for the AVRISP mkll programmer, which you can't be using since it's for AVR devices. You need to select "Tools->Programmer->Arduino101" for burning bootloader.

noelpaz, 2016-10-18:
@az51911  - What you selected in the programmer will not have an impact to you uploading the sketch. If you choose the Arduino 101 if only updates a specific portion of the chip.

Having said that are you having problems with the default 1.0.7 to upload a sketch or are you actually trying out the latest version of https://github.com/01org/corelibs-arduino101.

If that is the case the arduino101load you just created using go build will have to be copied over to this location: 

.arduino15/packages/Intel/tools/arduino101load/1.6.9+1.28/arduino101load/ . I suggest making a backup of the original so you can go back safetly.

It is also possible that file corruption has occurred so reinstalling 1.0.7  gain is not a bad idea.

az51911, 2016-10-18:
@noelpaz you are right, when I put the arduino101load into that folder you specified, I no longer have the error. However...even though it says the sketch has uploaded...the LED isn't blinking.

Sketch uses 48,004 bytes (30%) of program storage space. Maximum is 155,648 bytes.
Starting download script...
Flashing is taking longer than expected
Try pressing MASTER_RESET button
dfu-util: Cannot open DFU device 8087:0aba
SUCCESS: Sketch will execute in about 5 seconds.

noelpaz, 2016-10-19:
@az51911 ... If it is the CureiBLE LED example, you have to have a central device like an  Android with BLE Version 5.0 and above and then download the nrfController app. Then using this app, connect to the LED peripheral device you need to write to the characteristic a value other the 0x00 and the the LED will light up. If you write a value of 0x00 then it will turn off. In iOS (4s and above)  LightBlue is the free BLE app. If you have 2 Arduino101. You can upload the LEDCentral example to one and the LED to another. You will see the led blink on both. LED Central updates the characteristic every 2 - 3 seconds.

doctim77, 2016-10-19:
I am unable to upload a sketch to the board using BLE central mode.  I continue to get the following error: 
!! BLE firmware version is not in sync with CurieBLE library !!. I then successfully flash the board, but nonetheless I still cannot upload a sketch and get the same error telling me that the firmware is not in sync.  I have tried this dozens of times with no success.

This is following updating to the most recent corelibs-arduino101 repo. 

Running the most current Arduino IDE version 1.6.12

I am desperately in need of BLE central functionality for a project that I am working on.  Anyone have some thoughts on how to remedy this issue? Thank you greatly for any help you can provide.

facchinm, 2016-10-19:
To use the mainline `corelibs-arduino101` core, you need to flash the ble_flashpack_v3 from here https://github.com/01org/corelibs-arduino101/releases/tag/2.0.0 . 
This needs to be done once, externally (eg. not using `Burn Bootloader` feature). 
It will be integrated once 2.0.0 is released. If you decide to go back to 1.0.7 you need to reflash the "older" ble firmware via `Burn booloader` menu

claudiaAl, 2016-10-19:
@facchinm , can you please provide details on how to flash the new ble firmware? I'm new to this and am also attempting to use the most recent code in the repo.  Thanks!

eriknyquist, 2016-10-24:
@az51911 Can you please try the method suggested by @facchinm and let us know if it worked?

az51911, 2016-10-24:
@eriknyquist  Yes I tried it, I was able to load the program (according to the Arduino sketch IDE), but nothing was manifested on the board. So I switched back to 1.0.7 at the moment since that is working with no issues.

eriknyquist, 2016-10-24:
OK. Since 2.0.0 is not actually released yet (i.e. you are playing with currently unsupported code), I recommend to close this issue. If you are still having this issue with the release package, then we can re-open it.

closed 2016-10-24 by kitsunami