arduino/ArduinoCore-arc32 #332: Arduino 101 can use u8g library?

vyuke, 2016-10-24:
Hi，i want my oled drive by arduino 101,compilation is successful,connect is ok,but OLED  doesn't work,i don't know why,please help me find the problem,thanks.
my code:
`#include "U8glib.h"

#define REST 4
#define DC 5

U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NONE);

void draw(void) {
  // graphic commands to redraw the complete screen should be placed here  
  u8g.setFont(u8g_font_unifont);
  //u8g.setFont(u8g_font_osb21);
  u8g.drawStr( 20, 36, "OPENJUMPER");
}

void setup(void) {
  pinMode(REST,OUTPUT);
  pinMode(DC,OUTPUT);
  digitalWrite(REST,HIGH);
  digitalWrite(DC,LOW);
  
  // flip screen, if required
  // u8g.setRot180();
  
  // set SPI backup if required
  //u8g.setHardwareBackup(u8g_backup_avr_spi);

  // assign default color value
  if ( u8g.getMode() == U8G_MODE_R3G3B2 ) {
    u8g.setColorIndex(255);     // white
  }
  else if ( u8g.getMode() == U8G_MODE_GRAY2BIT ) {
    u8g.setColorIndex(3);         // max intensity
  }
  else if ( u8g.getMode() == U8G_MODE_BW ) {
    u8g.setColorIndex(1);         // pixel on
  }
  else if ( u8g.getMode() == U8G_MODE_HICOLOR ) {
    u8g.setHiColorByRGB(255,255,255);
  }
}

void loop(void) {
  // picture loop
  u8g.firstPage();  
  do {
    draw();
  } while( u8g.nextPage() );
  
  // rebuild the picture after some delay
  delay(500);
}`

eriknyquist, 2016-10-24:
Hi @vyuke, pleasse read our [guidelines for reportang an issue] (https://github.com/01org/corelibs-arduino101/wiki/Reporting-an-issue)

vyuke, 2016-10-27:
[image]

eriknyquist, 2016-10-27:
OK, the image is a start, but I still need some more information. Did you read this https://github.com/01org/corelibs-arduino101/wiki/Reporting-an-issue ? Please read it.

Specifically:

1. Provide an URL for the library you are using (I see multiple incarnations of the u8glib library)
2. Provide an URL for the OLED display you are using, so we can purchase it if we need to
3. Did you know this OLED display works? did you try it on another device? if so, which one?

vyuke, 2016-11-04:
yes ,i used this OLED on Arduino uno,is work。this OLED more informations URL is [http://openjumper.cn/12864-oled/](url)  and this OLED is ordinary 12864

vyuke, 2016-11-04:
I asked some about u8glig，a man commented me 101 will not work with U8glib. I have to use u8g2 (version 2 of u8glib)
https://github.com/olikraus/U8g2_Arduino
but the OLED does not work too

eriknyquist, 2016-11-04:
This library is for AVR only. It requires <avr/pgmspace.h>.

closed 2016-11-04 by eriknyquist