arduino/ArduinoCore-arc32 #334: analogWrite issue

agdl, 2016-10-25:
Hi, I found a bug ion the analogWrite function. Here the sample sketch I used to reproduce the issue:
```
int led = 3;           // the PWM pin the LED is attached to
int brightness = 0;    // how bright the LED is
int fadeAmount = 5;    // how many points to fade the LED by

// the setup routine runs once when you press reset:
void setup() {
  // declare pin 9 to be an output:
  pinMode(2, INPUT_PULLUP);
  pinMode(led, OUTPUT);
}

// the loop routine runs over and over again forever:
void loop() {
  // set the brightness of pin 9:
  analogWrite(led, brightness);

  // change the brightness for next time through the loop:
  brightness = brightness + fadeAmount;

  // reverse the direction of the fading at the ends of the fade:
  if (brightness <= 0 || brightness >= 255) {
    fadeAmount = -fadeAmount;
  }
  // wait for 30 milliseconds to see the dimming effect
  delay(30);

  if(digitalRead(2) == LOW){
    digitalWrite(led, LOW);
    while(true);
  }
}
```

Basically if you use a pin as PWM and then you use the ```digitalWrite(pin, LOW);``` the last PWM value stays on the pin.
Obviously writing ```analogWrite(pin, 0);``` the issue disappears

oqibidipo, 2016-10-25:
See #280.

kitsunami, 2016-10-31:
Duplicate to #280

closed 2016-10-31 by kitsunami